ipab-slmc/exotica #43: Remove old initialization

VladimirIvan, 2017-05-04:
There are three ways to initialize EXOTica classes in the old code:
 1. From XML: ```initDerived()``` for base classes.
 1. From JSON: ```reinitialise()``` for specific problems.
 1. Manual: ```initialiseManual``` for Scene.

These are now deprecated in favour of the new initialization. All references to old initialization should be removed.

cam586, 2017-05-08:
All deleted, compiles and runs examples

VladimirIvan, 2017-05-08:
There are couple more bits to remove:
 * ```Object::initBase()``` removing this will uncover all the hidden deprecated init calls
 * ```initBase()``` This is implemented in very major class
 * ```Factory::createObject()``` and its overloads (this will cause a lot of stuff to fail to compile initially)
 * ```initialise(const rapidjson::Value& a)``` in every task map
 * ```Scene::initialisation()```
 * ```KinematicTree::initKinematics()```
 * ```Server::initialise()```
 * ```Server::createParam()``` this parses XML as well
 * ```Server::registerParam()```
 * ```Server::registerRosParam()```
 * ```Server::registerStatic()```
 * ```Setup::listSolversAndProblems()```
 * ```Setup::initialise()``` and all overloads
 * ```Setup::initialiseProblemJSON()```
 * ```virtual void PlanningProblem::reinitialise()``` including implementations in specific problems

cam586, 2017-05-09:
The second batch has now all been removed (I think).

closed 2017-05-10 by VladimirIvan