ipab-slmc/exotica #57: Initializers don't work on Kinetic

wxmerkt, 2017-05-31:
The CMake macro logic for the initializers doesn't work with catkin tools if configured to _not_ use an install directory as the files are generated in the devel and not the source tree whereas ``${catkin_INCLUDE_DIRS}`` will return the source tree path. Generating into the source tree is not a good CMake option either. 

Questions thus are how to handle non-install configured workspaces?

[The issue arises by just building exotica - when building the examples it will fail as it cannot find the generated {something}Initializer.h file]

wxmerkt, 2017-05-31:
For instance: the ``IKSolverInitializer.h`` gets created in ``~/catkin_ws/devel/include/ik_solver/IKsolverInitializer.h``

While ``${catkin_INCLUDE_DIRS}`` in another project returns ``~/catkin_ws/src/exotica/exotations/solvers/ik_solver/include``

wxmerkt, 2017-05-31:
@VladimirIvan: This issue affects **all non-install workspaces** (which is what we're currently recommending for development, and also for OH). 
I've found a temporary fix for this issue, cf. https://github.com/openhumanoids/exotica/commit/98aa046a3b8fc83b3a926b3d1ba9c55aa7aeef89

However, this is *not* how this should be resolved. The new initialisation creates files header files without exported targets (cross package that is) - these are correctly placed into the devel tree and then installed into the install tree. When running ``find_package(catkin COMPONENTS ...)``, however, it will set up ``{package}_INCLUDE_DIRS`` with the source-tree include path as defined in ``catkin_package(INCLUDE )`` excluding the generated file tree (in DEVEL_PREFIX/include). We should not, on the other hand, create those generated files in the source tree either (which may be an alternate quick fix/workaround). 

I'm not sure how to properly resolve this right now but wanted to flag it up. Perhaps the above fix is OK for now (though potentially unclean if someone uses both install and devel spaces) and we should keep track of it until a better solution has been found.

wxmerkt, 2017-06-01:
I found the proper fix to this issue and included it as part of #58

VladimirIvan, 2017-06-04:
I'm in favour of the proper fix. I'll have a better look at the changes you made next week. I did look into several ways to generate those files in Indigo but I was leaving upgrade to Kinetic for later (hence the issue coming up just now).

commit 380bc711 (2017-06-01): Proper fix to resolve #57
Note: initializers need to be added and generated BEFORE calling catkin_package
PR #58 (2017-06-16): Initializer fix and Compatibility with Ubuntu Xenial 16.04 LTS and Kinetic Kame
This PR brings in a range of updates:

1. It fixes the initializers for non-install spaces and thereby properly resolves #57 
2. It removes some non-required dependencies that were spewing warnings
3. It makes Exotica compatible with 16.04 and Kinetic Kame.

Fully tested with both ``catkin_make_isolated`` and ``catkin build`` on Xenial + Kinetic. Travis confirmed that all changes work with Trusty + Indigo.
closed 2017-06-16 by VladimirIvan