ipab-slmc/exotica #71: Python wrapper issue tracking

wxmerkt, 2017-07-29:
Running list of current issues with the Exotica Python wrapper (WIP):

- [ ] Python 3 support - should be by easy by setting ``-DPYBIND11_PYTHON_VERSION=3``, however, the libraries are not correctly picked up. I suspect this to be due to the non-standard why in which ``pybind/cmake/pybind.cmake`` is doing the python lookup
- [ ] Eigen version - requires >3.2.7 which is non-standard on 14.04 - need a way to deal with it
- [ ] The macro ``pybind_add_module`` should not use custom commands but the CMake native versions of creating folders and copying files; they also prevent system-specific issues
- [ ] Change submodule to externalproject if no changes
- [ ] Perhaps we should have a configure step to checkout header only externals such as pybind11 and Eigen to make sure we don't run into these issues?

Also, I would like to suggest to rename ``exotica_py`` to ``pyexotica``.

wxmerkt, 2017-08-02:
The segfault on exit stems from roscomm and is classified as a wont-fix for indigo, but is resolved in kinetic.

**References:**

- https://github.com/ros/ros_comm/pull/871
- https://github.com/ros/ros_comm/pull/1008

VladimirIvan, 2017-08-08:
1. This is a limitation of pybind11 and cmake's ability to find different versions of python. Unless we write a our own python version detector we won't be able to get around this. Using the path to executable is working for now.
1. No solution yet apart from getting system wide eigen >=3.2.7.
1. Fixed in #80
1. See #79 
1. for pybind11 see #79. Eigen has to be consistent across libraries. Best way to achieve that is to have a system install of eigen as far as I can tell.

PR #84 (2017-08-09): Update README to include Eigen >3.2.7 requirement
Resolves #71
commit ceb2b5af (2017-08-09): Update README to include Eigen >3.2.7 requirement

Resolves #71
closed 2017-08-09 by VladimirIvan