ipab-slmc/exotica #96: KinematicTree/Kinematica uses hard-coded joint limits

wxmerkt, 2017-09-04:
Cf. https://github.com/openhumanoids/exotica/blob/master/exotica/src/KinematicTree.cpp#L434-L454

VladimirIvan, 2017-09-05:
How is this handled in MoveIt? Are these specified somewhere in URDF/SRDF or in a additional YAML file?
It would still be good to have a default hard coded value. Maybe with more round numbers.
There is a method for setting these at runtime. The values have to be specified for OMPL to work with floating joints so we need some way to ensure the joint limits are always present.

wxmerkt, 2017-09-09:
Turns out there was a bug here as well - the joint limits of the first 3 or 6 controlled joints were overwritten if the model had a floating base but the controlled group did not. Fix is in #109.

wxmerkt, 2018-12-11:
You can set the floating base joint limits in Python, which is what I resort to automatically. You may notice that the particularly chosen values resemble the planning values on _a certain humanoid_. 

Here are the methods for setting the floating base limits (and all of them, if the floating base is actuated):

https://github.com/ipab-slmc/exotica/blob/d446890f49f7b783bcf0004799578cf9d34fd222/exotica/include/exotica/KinematicTree.h#L162-L164

These are exposed to Python (and I use them heavily).

christian-rauch, 2018-12-11:
I already figured that these values fit well to a particular humanoid robot :-)

I got apparent wrong results using the `JointLimit` task before and therefore never used it. Just now that I searched the source code, I was able to retrace the results.

Because potential future users of exotica may hit similar issues (get weird results) which are hard to debug (searching the source code) I would basically propose to not limit the floating base joints by default.

Also, I disapprove hard coded values in general :-)

christian-rauch, 2018-12-14:
I am closing this (for good?) since the joint limits have been set to unbound by default in PR https://github.com/ipab-slmc/exotica/pull/435.

commit 0954c04b (2017-09-09): KinematicTree: Fix wrong joint limits for ControlledBaseType
- Relates to #105 and #96

Resolves #96
closed 2018-12-14 by christian-rauch