ipab-slmc/exotica #100: ROS messages passing through Python wrapper

wxmerkt, 2017-09-05:
As discussed via WhatsApp yesterday, we need a way to pass ROS messages through the Python wrapper to Python-wrapped functions. We are currently investigating the use of serialization/deserialization of messages through buffers, however, they currently segfault at the conversion on the C++ side. Additionally, ``StringIO`` isn't available in Python3 anymore: http://python-future.org/compatible_idioms.html#stringio

VladimirIvan, 2017-09-05:
The rest of the wrapper works with python3 so I have no objection to making it fully compatible on the pyexotica side.
That said, ROS does not officially support python3 so creating the messages may require extra work.

wxmerkt, 2017-09-05:
I've now added Python3 compatibility as part of #99

PR #99 (2017-09-05): Collision and planning scene updates and fixes
- Adds ``getClosestDistance`` to ``CollisionScene`` with MoveIt collision checking and wraps it for Python (through ``Scene``)
- Removes ``self`` argument from ``isStateValid(q)`` overload as unused (a step towards simplifying #39) - it was unused in all online exotica-based projects
- Unified ``debug_`` and ``visual_debug_`` as the latter couldn't be enabled, ``debug_`` can be set interactively from Python (``debugMode=True``)
- Exposes ``publishScene``
- Adds wrapping for ROS message types (should work for both Python 2 and 3, only tested with P [... cut]
closed 2017-09-05 by wxmerkt