ipab-slmc/exotica #134: Bug in CoM task / mass extraction

wxmerkt, 2017-09-30:
Working on a CoM stability criterion and noticed that the Valkyrie mass came in at **127.686 kg** when we usually (e.g. with Drake) get **135.906 kg**. Went to print out the individual link masses and they are off (perhaps an off by one bug?):

```
Link 0 (world_frame): 0 kg
Link 1 (world_joint/trans_y): 0 kg
Link 2 (world_joint/trans_z): 0 kg
Link 3 (world_joint/rot_x): 0 kg
Link 4 (world_joint/rot_y): 0 kg
Link 5 (world_joint/rot_z): 0 kg
Link 6 (pelvis): 2.39 kg
Link 7 (leftHipYawLink): 3.665 kg
Link 8 (leftHipRollLink): 10.2 kg
Link 9 (leftHipPitchLink): 6.2 kg
Link 10 (leftKneePitchLink): 0.03 kg
Link 11 (leftAnklePitchLink): 2.27 kg
Link 12 (leftFoot): 0.1 kg
Link 13 (pelvis): 1e-07 kg
Link 14 (pelvis): 1e-07 kg
Link 15 (pelvis): 2.39 kg
Link 16 (rightHipYawLink): 3.665 kg
Link 17 (rightHipRollLink): 10.2 kg
Link 18 (rightHipPitchLink): 6.2 kg
Link 19 (rightKneePitchLink): 0.03 kg
Link 20 (rightAnklePitchLink): 2.27 kg
Link 21 (rightFoot): 0.1 kg
Link 22 (pelvis): 0.5 kg
Link 23 (torsoYawLink): 0.1 kg
Link 24 (torsoPitchLink): 39.47 kg
Link 25 (torso): 1e-07 kg
Link 26 (torso): 2.65 kg
Link 27 (leftShoulderPitchLink): 3.97 kg
Link 28 (leftShoulderRollLink): 3.085 kg
Link 29 (leftShoulderYawLink): 1.83 kg
Link 30 (leftElbowPitchLink): 2.476 kg
Link 31 (leftForearmLink): 0.14 kg
Link 32 (leftWristRollLink): 0.712 kg
Link 33 (leftPalm): 0.001 kg
Link 34 (l_hand_force_torque): 0.001 kg
Link 35 (l_hand_force_torque): 0.001 kg
Link 36 (leftPalm): 0.02 kg
Link 37 (leftIndexFingerPitch1Link): 0.018 kg
Link 38 (leftIndexFingerPitch2Link): 0.01 kg
Link 39 (leftPalm): 0.02 kg
Link 40 (leftMiddleFingerPitch1Link): 0.011 kg
Link 41 (leftMiddleFingerPitch2Link): 0.006 kg
Link 42 (leftPalm): 0.02 kg
Link 43 (leftPinkyPitch1Link): 0.011 kg
Link 44 (leftPinkyPitch2Link): 0.006 kg
Link 45 (leftPalm): 0.017 kg
Link 46 (leftThumbRollLink): 0.02 kg
Link 47 (leftThumbPitch1Link): 0.013 kg
Link 48 (leftThumbPitch2Link): 0.006 kg
Link 49 (torso): 1e-07 kg
Link 50 (torso): 1.05 kg
Link 51 (lowerNeckPitchLink): 1.4 kg
Link 52 (neckYawLink): 3.89 kg
Link 53 (upperNeckPitchLink): 1.41984 kg
Link 54 (head): 1e-05 kg
Link 55 (head): 1e-05 kg
Link 56 (head): 1e-05 kg
Link 57 (head): 0.057654 kg
Link 58 (hokuyo_link): 1e-05 kg
Link 59 (head): 1e-05 kg
Link 60 (left_camera_frame): 0 kg
Link 61 (head): 1e-05 kg
Link 62 (head): 1e-05 kg
Link 63 (right_camera_frame): 0 kg
Link 64 (head): 1e-05 kg
Link 65 (torso): 1e-07 kg
Link 66 (torso): 2.65 kg
Link 67 (rightShoulderPitchLink): 3.97 kg
Link 68 (rightShoulderRollLink): 3.085 kg
Link 69 (rightShoulderYawLink): 1.83 kg
Link 70 (rightElbowPitchLink): 2.476 kg
Link 71 (rightForearmLink): 0.14 kg
Link 72 (rightWristRollLink): 0.712 kg
Link 73 (rightPalm): 0.001 kg
Link 74 (r_hand_force_torque): 0.001 kg
Link 75 (r_hand_force_torque): 0.001 kg
Link 76 (rightPalm): 0.02 kg
Link 77 (rightIndexFingerPitch1Link): 0.018 kg
Link 78 (rightIndexFingerPitch2Link): 0.01 kg
Link 79 (rightPalm): 0.02 kg
Link 80 (rightMiddleFingerPitch1Link): 0.011 kg
Link 81 (rightMiddleFingerPitch2Link): 0.006 kg
Link 82 (rightPalm): 0.02 kg
Link 83 (rightPinkyPitch1Link): 0.011 kg
Link 84 (rightPinkyPitch2Link): 0.006 kg
Link 85 (rightPalm): 0.017 kg
Link 86 (rightThumbRollLink): 0.02 kg
Link 87 (rightThumbPitch1Link): 0.013 kg
Link 88 (rightThumbPitch2Link): 0.006 kg
```

For reference, ``pelvis`` should be **8.22**.

wxmerkt, 2017-09-30:
I think I figured out the cause - the root link which is the pelvis isn't calculated into the overall mass (KDL doesn't support root link inertia) and that's exactly the difference

wxmerkt, 2017-10-01:
Re-opening as this causes the calculated center of mass to be wrong even if an artificial link is added. Working on the proper fix by re-adding the inertial of the root link.

Drake 135kg CoM with zeros(38,1): [-0.0080, -0.0008, -0.0239]
Exotica 127kg CoM with zeros(38,1): [-0.00638323 -0.000615918   0.00591245]
Exotica 135kg CoM with zeros(38,1): [-0.00631892 -0.000791083   0.00533711]

VladimirIvan, 2017-10-01:
Drake uses a separate URDF. Double check that some mass or offsets haven't been added to it.
Also check if there is precision loss. I don't think floats are used in drake for the model and the error should be around 1e-7 but I haven't worked with their code very much.

wxmerkt, 2017-10-01:
I loaded a RigidBodyManipulator from the same URDF; the same as I've always done it for dynamics experiments before. I verified the error lies with Exotica and I am close to having a fix. It may not be as easy for a floating base robot though as we are having six joints for the floating base and we'd need to make a decision where to assign the inertial to.

wxmerkt, 2017-10-01:
## Debug information as requested
- Both use the same URDF (valkyrie_sim_drake)
- Tested with ``zeros(38,1)`` and ``ones(38,1)``
- The Exotica results below are from the fixed URDF-tree that maintains the root inertial - commit 254dba0cc5579ffea222d21484ad504b3e825840

### Drake (Matlab)
```
world, 0, []
pelvis+pelvisMiddleImu_Frame+pelvisRearImu_Frame, 8.22, [-0.00532000079324695;-0.00351199991454988;-0.00360000257971777]
torsoYawLink, 0.5, [0;0;-0.01]
torsoPitchLink, 0.1, [0;0;0.005]
torso+leftHazardCamera_Frame+rightHazardCamera_Frame+leftTorsoImu_Frame, 39.47, [-0.0955479992579665;-0.00333699963453256;0.24309799964726]
lowerNeckPitchLink, 1.05, [-0.02;0;0.04]
neckYawLink, 1.4, [-0.03;-0.01;0.15]
upperNeckPitchLink+head+head_imu_link+right_led_frame+left_led_frame+center_bottom_led_frame+center_top_led_frame+right_camera_frame+right_camera_optical_frame+left_camera_frame+left_camera_optical_frame+hokuyo_link+head_hokuyo_frame, 5.3676, [0.102665331720606;0.00417774135171425;0.0404094265796232]
rightShoulderPitchLink, 2.65, [0.012;-0.251;0]
rightShoulderRollLink, 3.97, [0.008513;-0.02068;-0.001088]
rightShoulderYawLink, 3.085, [-0.004304;-0.209832;-0.007295]
rightElbowPitchLink, 1.83, [-0.020344;-0.014722;-0.0223]
rightForearmLink, 2.476, [0.015;-0.13;0.019564]
rightWristRollLink, 0.14, [0;0;0]
rightPalm+r_hand_force_torque+r_hand_face+right_palm+rightPinkyPitch1Link+rightPinkyPitch2Link+rightPinkyPitch3Link+rightMiddleFingerPitch1Link+rightMiddleFingerPitch2Link+rightMiddleFingerPitch3Link+rightIndexFingerPitch1Link+rightIndexFingerPitch2Link+rightIndexFingerPitch3Link+rightThumbRollLink+rightThumbPitch1Link+rightThumbPitch2Link+rightThumbPitch3Link, 0.893, [0.00301349160134455;-0.0626843399962243;0.00319254213199124]
leftShoulderPitchLink, 2.65, [-0.012;0.251;0]
leftShoulderRollLink, 3.97, [-0.008513;0.02068;-0.001088]
leftShoulderYawLink, 3.085, [-0.004304;0.209832;0.007295]
leftElbowPitchLink, 1.83, [-0.020344;0.014722;0.0223]
leftForearmLink, 2.476, [0.015;0.13;0.019564]
leftWristRollLink, 0.14, [0;0;0]
leftPalm+l_hand_force_torque+l_hand_face+left_palm+leftPinkyPitch1Link+leftPinkyPitch2Link+leftPinkyPitch3Link+leftMiddleFingerPitch1Link+leftMiddleFingerPitch2Link+leftMiddleFingerPitch3Link+leftIndexFingerPitch1Link+leftIndexFingerPitch2Link+leftIndexFingerPitch3Link+leftThumbRollLink+leftThumbPitch1Link+leftThumbPitch2Link+leftThumbPitch3Link, 0.893, [0.00301349160134455;0.0626843399962243;0.00319254213199074]
rightHipYawLink, 2.39, [0.02176;0.00131;0.03867]
rightHipRollLink, 3.665, [0.012959;-0.00755;-0.01595]
rightHipPitchLink, 10.2, [0.016691;-0.091397;-0.207875]
rightKneePitchLink, 6.2, [-0.022183;0.001703;-0.189418]
rightAnklePitchLink, 0.03, [0;0;0]
rightFoot+rightCOP_Frame, 2.37, [0.0369086666666667;0.00494323628691983;-0.0489279240506329]
leftHipYawLink, 2.39, [0.02176;-0.00131;0.03867]
leftHipRollLink, 3.665, [0.012959;0.00755;-0.01595]
leftHipPitchLink, 10.2, [0.016691;0.091397;-0.207875]
leftKneePitchLink, 6.2, [-0.022183;0.001703;-0.189418]
leftAnklePitchLink, 0.03, [0;0;0]
leftFoot+leftCOP_Frame, 2.37, [0.0369086666666667;0.00494323628691983;-0.0489279240506329]
```

### Exotica (with fixed root inertial)
```
world_frame, mass: 0, CoM: [           0,           0,           0]
world_joint/trans_y, mass: 0, CoM: [           0,           0,           0]
world_joint/trans_z, mass: 0, CoM: [           0,           0,           0]
world_joint/rot_x, mass: 0, CoM: [           0,           0,           0]
world_joint/rot_y, mass: 0, CoM: [           0,           0,           0]
world_joint/rot_z, mass: 0, CoM: [           0,           0,           0]
pelvis, mass: 8.22, CoM: [    -0.00532,   -0.003512,     -0.0036]
leftHipYawLink, mass: 2.39, CoM: [     0.02176,    -0.00131,     0.03867]
leftHipRollLink, mass: 3.665, CoM: [    0.012959,     0.00755,    -0.01595]
leftHipPitchLink, mass: 10.2, CoM: [    0.016691,    0.091397,   -0.207875]
leftKneePitchLink, mass: 6.2, CoM: [   -0.022183,    0.001703,   -0.189418]
leftAnklePitchLink, mass: 0.03, CoM: [           0,           0,           0]
leftFoot, mass: 2.27, CoM: [    0.037702,    0.005161,   -0.047634]
leftCOP_Frame, mass: 0.1, CoM: [           0,           0,           0]
pelvisMiddleImu_Frame, mass: 1e-07, CoM: [           0,           0,           0]
pelvisRearImu_Frame, mass: 1e-07, CoM: [           0,           0,           0]
rightHipYawLink, mass: 2.39, CoM: [     0.02176,     0.00131,     0.03867]
rightHipRollLink, mass: 3.665, CoM: [    0.012959,    -0.00755,    -0.01595]
rightHipPitchLink, mass: 10.2, CoM: [    0.016691,   -0.091397,   -0.207875]
rightKneePitchLink, mass: 6.2, CoM: [   -0.022183,    0.001703,   -0.189418]
rightAnklePitchLink, mass: 0.03, CoM: [           0,           0,           0]
rightFoot, mass: 2.27, CoM: [    0.037702,    0.005161,   -0.047634]
rightCOP_Frame, mass: 0.1, CoM: [           0,           0,           0]
torsoYawLink, mass: 0.5, CoM: [           0,           0,       -0.01]
torsoPitchLink, mass: 0.1, CoM: [           0,           0,       0.005]
[... 41 lines cut ...]
rightHazardCamera_Frame, mass: 1e-07, CoM: [           0,           0,           0]
rightShoulderPitchLink, mass: 2.65, CoM: [       0.012,      -0.251,           0]
rightShoulderRollLink, mass: 3.97, CoM: [    0.008513,    -0.02068,   -0.001088]
rightShoulderYawLink, mass: 3.085, CoM: [   -0.004304,   -0.209832,   -0.007295]
rightElbowPitchLink, mass: 1.83, CoM: [   -0.020344,   -0.014722,     -0.0223]
[... cut ...]
rightPinkyPitch1Link, mass: 0.02, CoM: [           0,       -0.02,           0]
rightPinkyPitch2Link, mass: 0.011, CoM: [           0,      -0.013,           0]
rightPinkyPitch3Link, mass: 0.006, CoM: [           0,       -0.01,           0]
rightThumbRollLink, mass: 0.017, CoM: [           0,          -0,        0.01]
rightThumbPitch1Link, mass: 0.02, CoM: [           0,      -0.005,        0.02]
rightThumbPitch2Link, mass: 0.013, CoM: [           0,      -0.004,       0.017]
[EXOTica]: [CoM] Total model mass: 135.906 kg
```

commit 2b12ae04 (2017-10-01): CoM: Fix CoM computation bug, fixes #134
The calculation of the Centre of Gravity of links was wrong as the relative frame was expressed in the parent rather than the child link's frame.
commit f671deea (2017-10-02): CoM: Fix CoM computation bug, fixes #134
The calculation of the Centre of Gravity of links was wrong as the relative frame was expressed in the parent rather than the child link's frame.
PR #137 (2017-10-03): Fix root inertial for floating-base robots and Center of Mass computation bug, fixes trans_x not existing, fixes floating base rotation
Resolves #134 
Resolves #136 
Resolves #138 

- Fixes extraction and assignment of root inertials, i.e. the mass and Center of Gravity of the root link which KDL neglects
- Fixes advertising & publishing of debug markers for the CoM task
- Fixes wrong center of mass calculation due to wrong relative frame/transform - verified against Drake for an all-zero configuration -- bug got introduced in https://github.com/ipab-slmc/exotica/commit/e89564c1647f742c236520b [... cut]
closed 2017-10-03 by wxmerkt