ipab-slmc/exotica #143: OMPL restructuring

VladimirIvan, 2017-10-09:
- Crate a library exposing common OMPL features, e.g. state checking, state conversion, solver setup, ...
- Remove two stage OMPL solver plugin - no more `OMPLimpl`.
- Implement OMPL solvers individually: RRT, RRTConnect, RRT*, BKPIECE, ...
- Each solver will link to the common library.
- Each solver can then use its own initializer and specific problem.

yimingyanged, 2017-10-10:
So, we want to have everything working after this restructuring (fixed-base arm, humanoid, time-indexed, etc)
- all OMPL algorithms should be able to solve different problems (state spaces)
- all problems also should be solvable by different algorithms, and we have the following state spaces
-- real vector state space
-- real vector and floating base state space
-- real vector and time state space
- also, we want to have the ability to add new solvers and state spaces freely in the future without changing the existing code

VladimirIvan, 2017-10-10:
No.
- There should be 3 state spaces that deal with the base type: fixed, mobile, floating. Each state space will be created based on what base type is used in the scene: ```KinematicTree::getControlledBaseType()```.
- Every OMPL algorithm should be able handle these three state spaces.
- Each OMPL algorithm can only solve one type of problem.
- One problem can be solved by multiple solvers.
- ```TimeIndexedSamplingProblem``` will add a time indexed version of fixed, mobile and floating state space. 
- A dedicated solver will be implemented for the ```TimeIndexedSamplingProblem```.
- Common tools should be implemented in a OMPL interface library. This library should not be plugin. Just a regular library that gets exported. All OMPL based solvers will use this labrary to handle state conversion, state space setup, state validation, ...

yimingyanged, 2017-10-13:
#148

closed 2017-10-18 by yimingyanged