ipab-slmc/exotica #153: Calculated distance points for meshes wrong for large triangles

wxmerkt, 2017-10-16:
This is most likely an issue with FCL or the way we are pushing meshes from the MoveIt planning scene to FCL (visually, the faces do not show up). The CollisionProxy does not point to the closest point...

[image]

wxmerkt, 2017-10-16:
Actually, this worked better with the old FCL before the big CollisionScene re-shuffling

VladimirIvan, 2017-10-17:
I've seen this issue with shape primitives as well. I switched to the ```fcl::GST_INDEP``` solver to fix it. That may not be a good solver for meshes?
Do small triangles work fine?

wxmerkt, 2017-10-17:
This has been fixed in #159 as well:

[image]

closed 2017-10-17 by wxmerkt