ipab-slmc/exotica #155: Collision Checking with Meshes Broken

wxmerkt, 2017-10-16:
Just noticed that with the new collision scene OMPL is broken for a scene with meshes (the same I have been using for weeks) - i.e. RRT plans a straight line through the mesh as if it wasn't there. This essentially breaks all our collision checking against meshes.

Even for the normal IK scene when I visually cannot see any collision in neither the planning scene nor the published collision shapes it returns 943 collision proxies with ~650 being closer than a centimetre.

This is a more general case of #153 - not only the points are wrong, the entire collision check is broken (both for old and new FCL).

wxmerkt, 2017-10-16:
Actually, how is this even possible?

```
Proxy: 'leftKneePitchLink_collision_0' - 'rightHipPitchLink_collision_0', c1: 0.442919 0.337705 0.398604 c1:   0.24 0.0984 0.7788 d: 0
```

for:

[image]

VladimirIvan, 2017-10-17:
I didn't test this with meshes.
FCL issues and PRs would suggest that not all their algorithms are fully implemented for all shapes yet. Check if ```fcl::GST_LIBCCD``` and ```fcl::GST_INDEP``` have the same issue.

PR #159 (2017-10-17): Fix collision checking: change to INDEP, update object transforms when in lazy checking mode
- Passes information on whether the collision scene is updated lazily to the CollisionScene
- Uses that information to force an object transform update when queried and it has not been externally updated
- This is e.g. of relevance when using multiple task maps that query the CollisionScene - in that case only one update would be required for multiple queries (and that update would be from AlwaysUpdateCollisionScene, i.e. forced from Scene)
- Switches to INDEP for collision queries in addition to dista [... cut]
closed 2017-10-17 by wxmerkt