ipab-slmc/exotica #184: Contact points for collision between primitive shape and mesh wrong

wxmerkt, 2017-11-03:
This is the proxy for the closest distance between the red hip yaw link sphere and the torso yaw link mesh:

[image]

VladimirIvan, 2017-11-03:
Could this be another local/global frame inconsistency based on colliding shape type?

wxmerkt, 2017-11-04:
Definitely, is a more complicated issue stemming from FCL. The contact points also change when swapping the order of e1 and e2, i.e. they are different for whether the primitive or the mesh is the first pair in the collision query.

I managed to fix and verify the contact points for most cases bar one. For primitive vs mesh I will add a work-around flipping the order internally.

###  Primitive vs Primitive: OK with INDEP, broken with LIBCCD
#### LIBCCD: broken
[image]

#### INDEP: OK
[image]

### Mesh vs Mesh: OK (with either INDEP and LIBCCD)
[image]

### Primitive vs Mesh: BROKEN
[image]

### Mesh vs Primitive: OK (with either INDEP and LIBCCD)
[image]

Aside, I found the following notes in the FCL code:
1. It is noted that INDEP is known not being able to compute the nearest points properly and we should rely on LIBCCD for that, cf. [here](https://github.com/flexible-collision-library/fcl/blob/master/test/test_fcl_signed_distance.cpp#L85-L86). I could not fully confirm that in my primitive-sphere vs primitive-mesh test and results were the same.
2. The nearest points are not guaranteed to lie on the surface of the objects when penetrating, cf. [here](https://github.com/flexible-collision-library/fcl/blob/master/include/fcl/narrowphase/distance-inl.h#L184-L185).
3. For meshes and octree it's not guaranteed to be on the surface of the object, cf. [here](https://github.com/flexible-collision-library/fcl/blob/master/include/fcl/narrowphase/distance-inl.h#L140-L146).

wxmerkt, 2017-11-06:
After more investigation we now need to consider 16 (!) different cases:

The four major pairs:

1. Primitive vs Primitive
2. Mesh vs Mesh
3. Primitive vs Mesh
4. Mesh vs Primitive

Two solvers:

1. INDEP
2. LIBCCD

Two main conditions:

1. In contact, i.e. closest distance < 0
2. Not in contact, i.e. closest distance > 0

And again each of these combinations comes with a different frame convention that needs to be respected when converting collision check results to global/world frame. I added a new commit implementing this in #190 and tested it on a complex concave (!) mesh vs primitive shapes and other complex concave meshes.

PR #190 (2017-11-06): Fix of nearest contact points and work-around for primitive-vs-mesh collisions
- Splits the collision types into four pairs and fixes contact points for each of the four cases.
- As highlighted in #184, primitive-vs-mesh can't be fixed right now. Thus, a work-around flipping the order internally to mesh-vs-primitive and changing return values is implemented. 
- Validated with both LIBCCD and INDEP for primitive-sphere vs mesh-sphere and mesh-cube.

However, while this works fine for the simple use cases it doesn't yet work with the more complex meshes used for experiments.

Resolves #184
closed 2017-11-06 by VladimirIvan