ipab-slmc/exotica #188: FCL latest broken on Kinetic

VladimirIvan, 2017-11-05:
FCL 0.6.0 crashes with segfault on Kinetic.
Default system installed FCL works fine.

This persists after #187.

Let's address this after sorting out #184.

wxmerkt, 2017-11-05:
Key here is that with self collisions enabled the check works, with self collisions disabled in an otherwise empty scene it crashes.

Stacktrace from running the unit test on Kinetic:

```
Program received signal SIGSEGV, Segmentation fault.
0x00007fffe0384261 in bool fcl::detail::setupMeshCollisionOrientedNode<fcl::OBBRSS<double>, fcl::detail::MeshCollisionTraversalNodeOBBRSS<double> >(fcl::detail::MeshCollisionTraversalNodeOBBRSS<double>&, fcl::BVHModel<fcl::OBBRSS<double> > const&, Eigen::Transform<fcl::OBBRSS<double>::S, 3, 18, 0> const&, fcl::BVHModel<fcl::OBBRSS<double> > const&, Eigen::Transform<fcl::OBBRSS<double>::S, 3, 18, 0> const&, fcl::CollisionRequest<fcl::OBBRSS<double>::S> const&, fcl::CollisionResult<fcl::OBBRSS<double>::S>&) () from /catkin_ws/devel/lib/libfcl.so.0.6
(gdb) backtrace
#0  0x00007fffe0384261 in bool fcl::detail::setupMeshCollisionOrientedNode<fcl::OBBRSS<double>, fcl::detail::MeshCollisionTraversalNodeOBBRSS<double> >(fcl::detail::MeshCollisionTraversalNodeOBBRSS<double>&, fcl::BVHModel<fcl::OBBRSS<double> > const&, Eigen::Transform<fcl::OBBRSS<double>::S, 3, 18, 0> const&, fcl::BVHModel<fcl::OBBRSS<double> > const&, Eigen::Transform<fcl::OBBRSS<double>::S, 3, 18, 0> const&, fcl::CollisionRequest<fcl::OBBRSS<double>::S> const&, fcl::CollisionResult<fcl::OBBRSS<double>::S>&) () from /catkin_ws/devel/lib/libfcl.so.0.6
#1  0x00007fffe006b702 in unsigned long fcl::detail::orientedMeshCollide<fcl::detail::MeshCollisionTraversalNodeOBBRSS<double>, fcl::OBBRSS<double> >(fcl::CollisionGeometry<fcl::OBBRSS<double>::S> const*, Eigen::Transform<fcl::OBBRSS<double>::S, 3, 18, 0> const&, fcl::CollisionGeometry<fcl::OBBRSS<double>::S> const*, Eigen::Transform<fcl::OBBRSS<double>::S, 3, 18, 0> const&, fcl::CollisionRequest<fcl::OBBRSS<double>::S> const&, fcl::CollisionResult<fcl::OBBRSS<double>::S>&) () from /catkin_ws/devel/lib/libfcl.so.0.6
#2  0x00007fffe0064e6c in unsigned long fcl::collide<double, fcl::detail::GJKSolver_indep<double> >(fcl::CollisionGeometry<double> const*, Eigen::Transform<double, 3, 18, 0> const&, fcl::CollisionGeometry<double> const*, Eigen::Transform<double, 3, 18, 0> const&, fcl::detail::GJKSolver_indep<double> const*, fcl::CollisionRequest<double> const&, fcl::CollisionResult<double>&) () from /catkin_ws/devel/lib/libfcl.so.0.6
#3  0x00007fffe0065b4a in unsigned long fcl::collide<double>(fcl::CollisionObject<double> const*, fcl::CollisionObject<double> const*, fcl::CollisionRequest<double> const&, fcl::CollisionResult<double>&) ()
   from /catkin_ws/devel/lib/libfcl.so.0.6
#4  0x00007fffe07c7e78 in exotica::CollisionSceneFCLLatest::checkCollision (o1=o1@entry=0x1ecd950, 
    o2=o2@entry=0x1ecdaa0, data=data@entry=0x7fffffffd920)
    at /catkin_ws/src/exotica/exotations/collision_scene_fcl_latest/src/CollisionSceneFCLLatest.cpp:226
#5  0x00007fffe07c82ce in exotica::CollisionSceneFCLLatest::collisionCallback (o1=0x1ecd950, o2=0x1ecdaa0, 
    data=0x7fffffffd920)
    at /catkin_ws/src/exotica/exotations/collision_scene_fcl_latest/src/CollisionSceneFCLLatest.cpp:244
#6  0x00007fffe040f422 in bool fcl::detail::dynamic_AABB_tree::collisionRecurse<double>(fcl::DynamicAABBTreeCollisionManager<double>::DynamicAABBNode*, fcl::DynamicAABBTreeCollisionManager<double>::DynamicAABBNode*, void*, bool (*)(fcl::CollisionObject<double>*, fcl::CollisionObject<double>*, void*)) () from /catkin_ws/devel/lib/libfcl.so.0.6
#7  0x00007fffe040f3e7 in bool fcl::detail::dynamic_AABB_tree::collisionRecurse<double>(fcl::DynamicAABBTreeCollisionManager<double>::DynamicAABBNode*, fcl::DynamicAABBTreeCollisionManager<double>::DynamicAABBNode*, void*, bool (*)(fcl::CollisionObject<double>*, fcl::CollisionObject<double>*, void*)) () from /catkin_ws/devel/lib/libfcl.so.0.6
#8  0x00007fffe040ff51 in fcl::DynamicAABBTreeCollisionManager<double>::collide(void*, bool (*)(fcl::CollisionObject<double>*, fcl::CollisionObject<double>*, void*)) const () from /catkin_ws/devel/lib/libfcl.so.0.6
#9  0x00007fffe07cb18f in exotica::CollisionSceneFCLLatest::isStateValid (this=0x189af30, self=<optimized out>, 
    safe_distance=0)
    at /catkin_ws/src/exotica/exotations/collision_scene_fcl_latest/src/CollisionSceneFCLLatest.cpp:323
```

wxmerkt, 2017-11-15:
With my upstream FCL bug fix for segfaults this is now reduced to a segfault when setting the transform:

```
Program received signal SIGSEGV, Segmentation fault.
0x00007fffdfef64a0 in fcl::CollisionObject<double>::setTransform(Eigen::Transform<double, 3, 18, 0> const&) ()
   from /catkin_ws/devel/lib/libfcl.so.0.6
(gdb) backtrace
#0  0x00007fffdfef64a0 in fcl::CollisionObject<double>::setTransform(Eigen::Transform<double, 3, 18, 0> const&) ()
   from /catkin_ws/devel/lib/libfcl.so.0.6
#1  0x00007fffe05662cf in exotica::CollisionSceneFCLLatest::updateCollisionObjectTransforms (this=0x18a3c80)
    at /catkin_ws/src/exotica/exotations/collision_scene_fcl_latest/src/CollisionSceneFCLLatest.cpp:90
#2  0x00007fffe0566a16 in exotica::CollisionSceneFCLLatest::isStateValid (this=0x18a3c80, self=<optimized out>, safe_distance=0)
    at /catkin_ws/src/exotica/exotations/collision_scene_fcl_latest/src/CollisionSceneFCLLatest.cpp:407
#3  0x00007ffff3aa216c in ?? () from /catkin_ws/devel/lib/python2.7/dist-packages/_pyexotica.so
#4  0x00007ffff3a8465e in ?? () from /catkin_ws/devel/lib/python2.7/dist-packages/_pyexotica.so
```

PR #198 (2017-11-15): Disable SSE alignment in fcl_catkin to avoid segmentation fault in Kinetic
After more debugging the segmentation fault for ``CollisionSceneFCLLatest`` could be traced back to an Eigen alignment issue during construction of the ``fcl::CollisionObjectd``. As the [default treatments](http://eigen.tuxfamily.org/dox-devel/group__TopicUnalignedArrayAssert.html) did not work out, I've decided to disable alignment/vectorisation for FCL. With this change the full test suite passes on Kinetic.

More information:

- http://eigen.tuxfamily.org/dox-devel/group__DenseMatrixManipulation__Alignement.html
- ht [... cut]
commit 9d6543ad (2017-11-15): Disable SSE alignment in fcl_catkin to avoid segmentation fault in Kinetic - resolves #188
closed 2017-11-15 by wxmerkt