ipab-slmc/exotica #224: Error with Python bindings

cmower, 2018-01-19:
Following installation tutorials [here](http://ipab-slmc.github.io/exotica/Installation.html). I encountered one warning:

```
Warnings   << pybind11_catkin:make /home/chris/Projects/exotica_testing_ws/logs/pybind11_catkin/build.make.000.log
Cloning into 'pybind11_src'...
Note: checking out '391c75447da13ef44cc2e119e005c0aeee442862'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by performing another checkout.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -b with the checkout command again. Example:

  git checkout -b <new-branch-name>

HEAD is now at 391c754... Update all remaining tests to new test styles
Cloning into 'tools/clang'...
cd /home/chris/Projects/exotica_testing_ws/build/pybind11_catkin; catkin build --get-env pybind11_catkin | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -

```

I tried launching the `Python*.launch` files, they do not work (all `Cpp*.launch` work as expected). Error message for `PythonPlanIK.launch`: 

```
Publishing IK
Traceback (most recent call last):
  File "/home/chris/Projects/exotica_testing_ws/src/exotica/examples/exotica_examples/scripts/example_ik.py", line 23, in <module>
    while not is_shutdown():
NameError: name 'is_shutdown' is not defined
[IKPlannerPythonExampleNode-2] process has died [pid 5676, exit code 1, cmd /home/chris/Projects/exotica_testing_ws/src/exotica/examples/exotica_examples/scripts/example_ik.py __name:=IKPlannerPythonExampleNode __log:=/home/chris/.ros/log/ad20eaec-fd41-11e7-821c-a0cec809263b/IKPlannerPythonExampleNode-2.log].
log file: /home/chris/.ros/log/ad20eaec-fd41-11e7-821c-a0cec809263b/IKPlannerPythonExampleNode-2*.log

```

I'm running ROS Kinetic on Ubuntu 16.04. No errors when @cam586 tested running ROS  Indigo on Ubuntu 14.04.

wxmerkt, 2018-01-19:
This is a file that hasn't been upgraded, i.e. the two of you aren't on the same branch/commit. We will fix it with the next merge (PR #220).

wxmerkt, 2018-01-19:
If you pull the new master this should be fixed.

cmower, 2018-01-19:
@wxmerkt yup, building in progress and `pybind11` succeeded. Will let you know when finished.

wxmerkt, 2018-01-19:
The warning is just a warning - nothing to worry about - and it has nothing to do with the bindings itself. The error above is that ``is_shutdown`` wasn't available anymore since we changed how Python scripts can exit gracefully (now with signal handlers). Examples have now been updated as well to use the new way of exiting.

closed 2018-01-19 by wxmerkt

cmower, 2018-01-19:
Thanks @wxmerkt. Build succeeded with no issues of any kind and Python examples working as expected.