ipab-slmc/exotica #239: Expose the LazyPRM OMPL planner in python

rfishy1, 2018-01-29:
The [OMPL LazyPRM planner](http://ompl.kavrakilab.org/classompl_1_1geometric_1_1LazyPRM.html#gLazyPRM) is not exposed in the python bindings for EXOTica.

Expanding the selection of exposed planners would improve the usefulness of EXOTica for benchmarking planning algorithms.

VladimirIvan, 2018-02-01:
I have added the LazyPRM but it's not able to solve even the sample problem. I don't think the OMPL implementation is working or we are missing a step in the planner setup.

@rfishy1 Have a look at 78b808f but I don't see an easy fix.

rfishy1, 2018-02-03:
@VladimirIvan Occasionally the solver works (about half the time for the example script), and uses between 1000 and 4000 problem updates. 

It seems that the ten second timeout is usually insufficient to solve the task (for a successful attempt). The image below has the solve times for the 10 loop iterations. It seems that once a path is found, the roadmap is retained and reused. I'll look at the setup to see whether there's a step which would improve the performance.

[image]

VladimirIvan, 2018-02-03:
I'll increase the the timeout in the example to 10s and merge then.

PR #255 (2018-02-03): Added LazyPRM
Resolves #239

The example has a fairly high failure rate.
closed 2018-02-03 by wxmerkt