ipab-slmc/exotica #259: Solver Wishlist

wxmerkt, 2018-02-03:
- [ ] [SNOPT](https://github.com/snopt/snopt-interface/)
- [ ] [IPOPT](https://www.coin-or.org/Ipopt/documentation/node10.html)
- [ ] [NLopt](https://github.com/stevengj/nlopt)
- [ ] [Gurobi](http://www.gurobi.com/documentation/7.5/refman.pdf)
- [ ] Levenberg-Marquardt (perhaps via [Eigen](https://eigen.tuxfamily.org/dox/unsupported/LevenbergMarquardt_2LevenbergMarquardt_8h_source.html))

wxmerkt, 2018-12-30:
Updates from recent implementations:

- Eigen's `LevenbergMarquardt` and `HybridNonLinearSolver` (dog-leg) are unsuitable: they have implicit assumptions we don't fulfil (e.g., dimension of value > variables)
- NLopt's unconstrained solvers are very fast, but constrained solvers do not behave properly on our problems :'(. Particularly, I cannot get the augmented Lagrangian to do anything useful, the SLSQP requires tuning and performs much worse than SNOPT as well (both exit early for different reasons). Note, NLopt also uses dense representations, i.e., will be slower for larger problems.

fjandrad, 2022-11-11:
We are looking to use SNOPT in the lab. But I do not see any implementation on exotica for using SNOPT. If you have it somewhere is it possible to gain access to it?