ipab-slmc/exotica #269: FCL segmentation fault upon querying distance

wxmerkt, 2018-03-05:
Over the past months we've seen randomly occurring segmentation faults when querying distances from CollisionSceneFCLLatest. So far it's been hard to debug them/track them down since they weren't easily reproducible. Here's a cause which we can reproduce and stays the same:

```
Compute distance for rightPalm_collision_1 and Shelf_collision_4
s1_pos(0.692234, 0.0309092, 1.21726), s1_quat(0.743296, -0.083652, -0.0229007, 0.663317,) 
s2_pos(0.8, 0, 0.735), s2_quat(0.707107, 0, 0, 0.707107)
```

And here's the traceback:

```
Compute distance for rightPalm_collision_1 and Shelf_collision_4s1_pos( 0.692234 0.0309092   1.21726), s1_quat(  0.743296  -0.083652 -0.0229007   0.663317,) & s2_pos(  0.8     0 0.735), s2_quat(0.707107        0        0 0.707107)

Program received signal SIGSEGV, Segmentation fault.
0x00007fffa2bcd221 in fcl::detail::GJKSolver_indep<double>::S fcl::distance<fcl::detail::GJKSolver_indep<double> >(fcl::CollisionGeometry<fcl::detail::GJKSolver_indep<double>::S> const*, Eigen::Transform<fcl::detail::GJKSolver_indep<double>::S, 3, 18, 0> const&, fcl::CollisionGeometry<fcl::detail::GJKSolver_indep<double>::S> const*, Eigen::Transform<fcl::detail::GJKSolver_indep<double>::S, 3, 18, 0> const&, fcl::detail::GJKSolver_indep<double> const*, fcl::DistanceRequest<fcl::detail::GJKSolver_indep<double>::S> const&, fcl::DistanceResult<fcl::detail::GJKSolver_indep<double>::S>&) () from exotica_ws/devel/lib/libfcl.so.0.6
(gdb) backtrace
#0  0x00007fffa2bcd221 in fcl::detail::GJKSolver_indep<double>::S fcl::distance<fcl::detail::GJKSolver_indep<double> >(fcl::CollisionGeometry<fcl::detail::GJKSolver_indep<double>::S> const*, Eigen::Transform<fcl::detail::GJKSolver_indep<double>::S, 3, 18, 0> const&, fcl::CollisionGeometry<fcl::detail::GJKSolver_indep<double>::S> const*, Eigen::Transform<fcl::detail::GJKSolver_indep<double>::S, 3, 18, 0> const&, fcl::detail::GJKSolver_indep<double> const*, fcl::DistanceRequest<fcl::detail::GJKSolver_indep<double>::S> const&, fcl::DistanceResult<fcl::detail::GJKSolver_indep<double>::S>&) () from exotica_ws/devel/lib/libfcl.so.0.6
#1  0x00007fffa2bceafd in double fcl::distance<double>(fcl::CollisionObject<double> const*, fcl::CollisionObject<double> const*, fcl::DistanceRequest<double> const&, fcl::DistanceResult<double>&) () from exotica_ws/devel/lib/libfcl.so.0.6
#2  0x00007fffb8390127 in exotica::CollisionSceneFCLLatest::computeDistance (o1=o1@entry=0x2a080e0, o2=o2@entry=0x2ddb100, 
    data=data@entry=0x7fffffffb380)
    at exotica_ws/src/exotica/exotations/collision_scene_fcl_latest/src/CollisionSceneFCLLatest.cpp:298
#3  0x00007fffb839158e in exotica::CollisionSceneFCLLatest::getCollisionDistance (this=0x29fe3f0, o1="rightPalm", 
    self=<optimised out>, disableCollisionSceneUpdate=<optimised out>)
    at exotica_ws/src/exotica/exotations/collision_scene_fcl_latest/src/CollisionSceneFCLLatest.cpp:577
#4  0x00007fffb8391d11 in exotica::CollisionSceneFCLLatest::getCollisionDistance (this=0x29fe3f0, objects=..., self=@0x2a0d0b9: true)
    at exotica_ws/src/exotica/exotations/collision_scene_fcl_latest/src/CollisionSceneFCLLatest.cpp:589
#5  0x00007fffdd768625 in exotica::SmoothCollisionDistance::update (this=0x2a0cfd0, x=..., phi=..., J=...)
    at exotica_ws/src/exotica/exotations/task_maps/task_map/src/SmoothCollisionDistance.cpp:56
#6  0x00007ffff1605a6c in exotica::UnconstrainedTimeIndexedProblem::Update (this=0x2abb4a0, x_in=..., t=t@entry=103)
    at exotica_ws/src/exotica/exotica/src/Problems/UnconstrainedTimeIndexedProblem.cpp:190
#7  0x00007fffe054cffe in exotica::AICOsolver::updateTaskMessage (this=this@entry=0x2b845d0, t=t@entry=103, qhat_t=..., 
    minimum_step_tolerance=minimum_step_tolerance@entry=0, maxStepSize=maxStepSize@entry=1)
    at exotica_ws/src/exotica/exotations/solvers/aico/src/AICOsolver.cpp:476
#8  0x00007fffe054ee9c in exotica::AICOsolver::updateTimeStep (this=this@entry=0x2b845d0, t=t@entry=103, 
    updateFwd=updateFwd@entry=true, updateBwd=updateBwd@entry=false, maxRelocationIterations=maxRelocationIterations@entry=1, 
    minimum_step_tolerance=1.0000000000000001e-05, forceRelocation=false, maxStepSize=maxStepSize@entry=1)
    at exotica_ws/src/exotica/exotations/solvers/aico/src/AICOsolver.cpp:529
#9  0x00007fffe054fefc in exotica::AICOsolver::step (this=this@entry=0x2b845d0)
    at exotica_ws/src/exotica/exotations/solvers/aico/src/AICOsolver.cpp:619
#10 0x00007fffe05517ce in exotica::AICOsolver::Solve (this=0x2b845d0, solution=...)
```

bmagyar, 2018-03-05:
Which FCL is this? I had similar issues with moveit, although thay was the
hierarchy tree. Issue was uninitialized iterators within FCL.

wxmerkt, 2018-03-05:
Latest master, I have seen flexible-collision-library/fcl#213 and trying the fix right now. Not sure it's the same cause though since we are using the newer FCL with methods introduced after 0.5.

wxmerkt, 2018-03-05:
The fix didn't work, however, the segmentation fault goes away when turning off ``-march=native`` - which is odd given it's a relatively recent i7.

closed 2018-03-09 by wxmerkt