ipab-slmc/exotica #299: Debugging generic "RuntimeError"

christian-rauch, 2018-05-01:
I get a `RuntimeError` without a error message when calling `solve()` on my instantiated solver in Python. However, this only happens with some problem configurations. I.e. I am generating the problem configuration (task maps and start state) dynamically at runtime in Python, and the `RuntimeError` only appears after setting the problem definition (on a newly instantiated solver) and solving for a couple of times.

The stack trace ends at `solve()`. How can I get more information about where the exception is raised. Have you seen exceptions without error messages before?
Alternatively, I run run the same problem in C++ to see if it is a Python binding issue or related to my input with the possibility to debug it easier. For this, I need a full XML of my planning problem. Is there a way to export my configuration into a XML file?

VladimirIvan, 2018-05-01:
> Have you seen exceptions without error messages before?
This happens when a generic exception is raised. All exotica exceptions should print a meaningful message including function names and line numbers. I haven't found a way to get generic exceptions to print anything more than whatever it already prints. The best way is to raise it within a c++ program and catch it with gdb. gdb on python directly won't help though, it has to be a c++ program.

> For this, I need a full XML of my planning problem. Is there a way to export my configuration into a XML file?
This would be a useful feature. You could get the Initializer you used to create the problem in python and turn it into a XML yourself. It won't include any changes you did to the problem/solver after creating it though. It would be cool to have a way to serialize the problem/solver/scene state.

christian-rauch, 2018-05-02:
After some debugging, I found that the exception is raised during matrix inversion at:
https://github.com/ipab-slmc/exotica/blob/8cb56600b85fce13457b94c276d9f4d1947108d9/exotations/solvers/ik_solver/src/ik_solver/IKSolver.cpp#L78

There might be a bug with streaming references into `throw_pretty`. After removing `<< A_` from the error message, and basically replacing it by:
`throw_pretty("Can't invert matrix. Cholesky decomposition failed: " << info);`
I indeed get a nice error message:
```
In [...]/exotica/exotations/solvers/ik_solver/src/ik_solver/IKSolver.cpp
inverseSymPosDef 81
[...] Can't invert matrix. Cholesky decomposition failed: 15
```

This might be similar to the issue https://github.com/ipab-slmc/exotica/issues/287 that I opened earlier.