ipab-slmc/exotica #340: exo.Setup.initRos() blocks SIGINT (Ctrl+C)

christian-rauch, 2018-06-26:
When using `exo.Setup.initRos()` (in python) I am unable to terminate the process via `Ctrl+C`, respectively `kill -s SIGINT <pid>`.

VladimirIvan, 2018-06-26:
Do the examples terminate?
Eg. `roslaunch exotica_examples PythonPlanIK.launch`

christian-rauch, 2018-06-26:
I checked `example_ik.py` which seems to solve the problem by registering its own signal handler via: `signal.signal(signal.SIGINT, sigIntHandler)`. After removing this signal handler, I observe the same behaviour (blocking with `initRos()`).

christian-rauch, 2018-06-26:
Is this a limitation for the Python wrapper only? I imageine registering a signal handler in the library (C++) it should do the same.
Do you know where the SIGINT is going? I think each process has a default handler unless its redefined. Is it catched and ignored by ROS?

VladimirIvan, 2018-09-25:
This is python behaviour only.
I believe that c++ handles this as an exception.

closed 2018-09-25 by VladimirIvan