ipab-slmc/exotica #360: TaskMap constructed but not deconstructed

christian-rauch, 2018-08-06:
When TaskMaps are created by `createMap()` (via the ClassLoader):
https://github.com/ipab-slmc/exotica/blob/5d7a273a662f0e60b157b031a61fb4ddf75db44a/exotica/include/exotica/Setup.h#L84-L89
they are created as shared pointers for the purpose of self managing their cleanup.

When using the same solver instance in pyexotica via `exo.Setup.createSolver` and reinitialising the TaskMaps multiple times via `exo.Setup.createProblem` I get TaskMaps that are constructed multiple times but never deconstructed. This in turn gives me problems with a 3rd party library that creates global objects. The library provides methods to remove these objects but since the deconstructor of the TaskMap never gets called, I have no way to know when the TaskMap is not used anymore.

I assume that there is no way to force the shared pointer to release and cleanup its managed object before creating new TaskMaps (of the same class) in `createMap()`.
As a workaround, I was thinking about providing a virtual method (e.g. `cleanup()`) that TaskMaps can implement to deconstruct their members.

Would this be a good solution to the problem? And where would this `cleanup()` method be called (e.g. storing the shared pointers on creation in a list and calling the `cleanup()` method at the next call to `createMap()`)?

VladimirIvan, 2018-08-07:
Shared pointers can't be force-deleted. Creating list of them will only create an additional reference but it won't allow you to actually destroy the instance it's pointing to.

This is probably caused by a reference loop (A holds a ptr of B and B holds a ptr of A). My guess would be that it's somewhere between problem/solver/scene and the TaskMap. The problem should own a `std::shared_ptr<exotica::TaskMap>` while everything else should only only use `std::weak_ptr<exotica::TaskMap>`. We resolved this already for the kinematics and collision checking. The TaskMap never caused a problem like this before.

Can you provide some code to reproduce this? It would make it easier to track down.

christian-rauch, 2018-08-31:
At the moment I am using a single instance of my taskmap, i.e. `Instantiate` gets only called once, and thus I am avoiding this issue. I will try to provide a minimal example later.

wxmerkt, 2021-02-08:
I realise this issue is 2.5 years old but I believe I found the fix for this behaviour. Do you happen to have a few pointers on how to reproduce/verify it easily?

christian-rauch, 2021-02-09:
Not really. I only experienced this issue as part of a more complex python script. I see that I wanted to provide a minimal example back then, but I never did probably because I couldn't reproduce this in a simpler setup.

Maybe this issue shows already by using `exo.Setup.createProblem` multiple times on the same tasks? I think the problem was basically that this way, a task's deconstructor was not called before the task was constructed again. So, if the task's deconstructor gets called now, everything should be fine and we can close this issue.

wxmerkt, 2021-02-09:
Thanks! Yes, there was an issue with missing destructors for tasks. I fixed that in #731 so I'll close the issue for now. We can reopen if it reoccurs.

closed 2021-02-09 by wxmerkt