ipab-slmc/exotica #384: Access TaskMaps at runtime

christian-rauch, 2018-10-08:
I would like to access TaskMaps during runtime to set/get attributes and change its behaviour. Is there a way in C++ to get a reference or pointer to a specific taskmap?

Alternatively, I can imagine that getting/setting of properties can be achieved by using `Initializer`. Would there be interest in such a feature, i.e. adding getting and setting methods to TaskMaps?

VladimirIvan, 2018-10-11:
'PlanningProblem::getTaskMaps()' in c++ will return all the TaskMaps as a map. ''PlanningProblem::getTasks()' will return a vector of the TaskMaps. The latter is exposed in python as well. You have to cast the TaskMap to the specific type and you can then call TaskMap-specific functions on it. Python does the casting for you. You have to include the include 'pyexotica' first, then include 'task_map_py' (subject to some warnings that may need fixing).

Rho and Goal can be set from the problem, no need to get down to the TaskMap.

wxmerkt, 2018-10-22:
@christian-rauch is this resolved/can this be closed?

christian-rauch, 2018-10-22:
I managed to set properties of the task map via the proposed python wrapper. Using the python wrapper is quite efficient since I do not need to copy data, i.e. my numpy and opencv matrices point to the same data in memory.

However, this creates some boiler plate code for every method and task.
Would it be possible to use a virtual methods (get/set) and the de/-serialisation from the `exotica::Initializer`? I assume something like call by pointer or reference would not be possible in this case?

closed 2018-10-30 by christian-rauch