ipab-slmc/exotica #391: Implementation of PseudoInverse

christian-rauch, 2018-10-31:
Which algorithm is the basis for `IKsolver::PseudoInverse`?

In the configuration J.cols() < J.rows() and C==0 the pseudo inverse is computed as:
```C++
Jpinv = inverseSymPosDef(J.transpose() * J + W) * J.transpose();
```

For  W=I (identity) is this supposed to provide the same result as the Moore-Penrose pseudoinverse?

I have a case where `IKsolver` gives me quite small update steps (q_d) for a sparse Jacobian, while Matlab's `pinv()` seems to provide the correct q_d.

christian-rauch, 2018-11-01:
Relacing 
```C++
Eigen::MatrixXd Jinv = PseudoInverse(prob_->Cost.S * prob_->Cost.J);
```
with 
```C++
Eigen::MatrixXd Jinv = (prob_->Cost.S*prob_->Cost.J).completeOrthogonalDecomposition().pseudoInverse();
```
already gives me a faster convergence.
How would I need to set `C` and `W` to obtain the same pseudo inverse from `IKsolver::PseudoInverse` as provided by Eigen and Matlab?

christian-rauch, 2018-11-02:
Quick update: I implemented the Gauss-Newton (actually Levenberg-Marquardt) solver https://github.com/christian-rauch/gauss_newton_exo_solver to compare the convergence on a synthetic case. In the case shown, I am adding links iteratively to the set of optimised links, but this should not matter. Optimisation for 5 x 5 iterations:

Inverse Kinematics (IK):
[image]

Levenberg-Marquardt (LM):
[image]

The LM is using an adaptive damping factor with damping direction depending on the Hessian approximation as described here: https://uk.mathworks.com/help/optim/ug/least-squares-model-fitting-algorithms.html

VladimirIvan, 2018-11-02:
Thanks for pointing out the problems with the IK solver. I'll have a look at it and fix it.

I really like your minimal LM implementation. I'd love to add it to the core exotica.

PR #465 (2018-12-20): Changed the underlying method used by the IK solver
- Retired explicit pseudo-inverse computation, using a linear solver instead (solving for `(SJW)q=y`).
- Retired LAPACK, using `completeOrthogonalDecomposition` when available, `householderQr` otherwise.
- Runtime performance on both LWR and Valkyrie models is the same per single call.
- New solver is more numerically stable.
- Relates to #458 
- Fixes #391

Note:
- Setting C=0 means no regularisation. C=0.9999 (or similar) will regularise by virtually appending a scaled Identity matrix to the Jacobian. Values closer to 1 regularise more. The [... cut]
closed 2018-12-20 by wxmerkt