ipab-slmc/exotica #412: Eigen Alignment Issues

wxmerkt, 2018-11-27:
We are currently having severe Eigen alignment issues, at least on Ubuntu 16.04 with gcc (which is our de facto default setup). The underlying reason here is that 

(a) Ubuntu 16.04 ships a beta version of Eigen 3.3 by default.
(b) The default compiler is gcc 5.4.0.
(c) The AVX alignment here is broken (random segmentation faults).

**TODO:**

- Check our alignment and allocation operators.

Good resources:

- http://eigen.tuxfamily.org/dox/group__TopicStructHavingEigenMembers.html 
- https://github.com/RobotLocomotion/drake/issues/1854

VladimirIvan, 2018-11-28:
Is this different from #315 ?

wxmerkt, 2018-11-28:
It appears to be. #315 can be triggered with solely Exotica core and is related already to the 3.3-beta and could be fixed with workarounds. 

This one here is more explicitly to track required work to fix our allocators/alignments (which hopefully should close both at once).

closed 2018-11-30 by wxmerkt