ipab-slmc/exotica #426: ‘make_unique’ is not a member of ‘std’

christian-rauch, 2018-12-11:
When adding exotica as catkin component, I am unable to use `std::make_unique` (C++14).
E.g. this:
```
find_package(catkin REQUIRED COMPONENTS
  roscpp
  exotica
)
```
causes the compilation error: `error: ‘make_unique’ is not a member of ‘std’`.

While:
```
find_package(catkin REQUIRED COMPONENTS
  roscpp
  # exotica
)
```
compiles as expected.

See test catking package: [test_unique.zip](https://github.com/ipab-slmc/exotica/files/2667407/test_unique.zip) for full example.

This issue only appeared recently.

wxmerkt, 2018-12-11:
We are currently exporting ``add_compile_options(-std=c++11)`` in our ``CFG_EXTRAS``. If you want to use C++14 style, you would have to either export it yourself or we can update our export definition. I realise Melodic is probably using C++14 while Kinetic is using C++11. Swapping exotica and roscpp in your find_package may also work.

christian-rauch, 2018-12-11:
I see that `add_compile_options(-std=c++11)` is added to `cmake/Exotica.cmake`. Is this intended?

I think this needs to be placed inside the `CMakeLists.txt` since you may want to compile exotica with C++11 but you do not want to tell projects that import exotica which C++ standard they have to use.

wxmerkt, 2018-12-11:
We do not restrict projects using EXOTica from using higher standards - it is intended as a minimum as far as dependent projects are concerned that depend on EXOTica headers. Just add ``add_compile_options(-std=c++14)`` to your project.

christian-rauch, 2018-12-11:
But the problem is, that I added `add_compile_options(-std=c++14)` to my project (see the test project) but it seems that it gets overwritten by exotica's `add_compile_options(-std=c++11)`.

closed 2018-12-12 by wxmerkt