ipab-slmc/exotica #436: SIGSEGV with JointLimit taskmap

christian-rauch, 2018-12-12:
I am getting SIGSEGV at different random locations (libexotica, python, class_loader, libmoveit_collision_detection, ...) when using the `JointLimit` taskmap.

Running the example [sample_minimal.py.txt](https://github.com/ipab-slmc/exotica/files/2673242/sample_minimal.py.txt) with exotica on master (d446890f49f7b783bcf0004799578cf9d34fd222) results in a SIGSEGV after some loops.

With solver `IKSolver`, the returned solution have the form:
```
sol [-6.20736784e-311 -2.56435246e-312 -6.20736784e-311  4.82084560e-311
 -6.20736784e-311 -9.27114570e-311 -6.20736784e-311]
```
The large exponents indicate that there is a random memory access that sooner or later results in SIGSEGV.

When I use another solver, e.g. `GNsolver`, I get solutions with reasonable values but still SIGSEGVs.

The example script is pretty self contained and only requires exotica and ROS. Can someone reproduce the issue?

christian-rauch, 2018-12-12:
The error does not occur if using the `JointLimit` task alone.

christian-rauch, 2018-12-12:
The apparently wrong values in the IKSolver were caused by myself (solved in https://github.com/ipab-slmc/exotica/pull/438). I now get reasonable values (like with other solvers), but it still SIGSEGVs.

wxmerkt, 2018-12-12:
The traceback I am getting is
```
#22 std::shared_ptr<exotica::UnconstrainedEndPoseProblem>::operator= (__r=..., this=0x555558334338)
    at /usr/include/c++/7/bits/shared_ptr.h:319
#23 exotica::IKSolver::specifyProblem (this=0x555558334260, 
    pointer=std::shared_ptr<exotica::PlanningProblem> (use count 6, weak count 1) = {...})
    at /home/wxm/dev/exotica_ws/src/exotica/exotations/solvers/exotica_ik_solver/src/ik_solver.cpp:83
#24 0x00007fffde15fe80 in ?? ()
```

I.e. in the ``static_pointer_cast`` of the specifyProblem it is failing. Is this a new issue with master that used to work before?

wxmerkt, 2018-12-12:
I've modified the default ``example_ik_noros.py`` to include a `JointLimit` task and even added a for-loop for creating and specifying problems (10k) - it did not crash. I can see the crash with your script though.

wxmerkt, 2018-12-12:
I tracked it down - it is *not* the JointLimit but the `EffOrientation` task in combination with using `Quaternion` that causes the issue. If you switch to `RPY` or any other type it works. 

cc: @VladimirIvan - Quaternion strikes again.

christian-rauch, 2018-12-12:
Good catch!
Is this something that might strike randomly with lower probability when not using the `JointLimit` task?
It appears I can run the script forever without `JointLimit` but with keeping the other tasks.

christian-rauch, 2018-12-12:
There is an error in the task setup. I am defining separate tasks for position and orientation (I used to have a single `EffFrame` task before). But I am setting a pose (3 position coefficients x,y,z + 4 quaternion coefficients w,x,y,z) for only the first task, which is the position task. The second orientation task will use default values.

But the error persists, if I replace the pos+ori task with a single `EffFrame` task: `('exotica/EffFrame', {'Name': "pose", 'Type': 'Quaternion', 'EndEffector': pose_frame}),`

PR #446 (2018-12-17): Fixes indexing of S matrix update
 - Fixes indexing of `S` in `updateS` of  `EndPoseTask`/`SamplingTask`/`TimeIndexedTask`.
 - Added testing of multiple cost terms to the task map tests (thus catching the above issue). This would have come back with a decent error message in the Debug build if this test was in place before.
 - Changed `eps` for tests using rotation types (AngleAxis now requires 1.1e-5).

Fixes #436
closed 2018-12-17 by VladimirIvan