ipab-slmc/exotica #441: example_ompl_noros.py fails

christian-rauch, 2018-12-13:
Running the example `example_ompl_noros.py` gives me:
```
python: /usr/include/eigen3/Eigen/src/Core/DenseBase.h:257: void Eigen::DenseBase<Derived>::resize(Eigen::Index, Eigen::Index) [with Derived = Eigen::Ref<Eigen::Matrix<double, -1, -1> >; Eigen::Index = long int]: Assertion `rows == this->rows() && cols == this->cols() && "DenseBase::resize() does not actually allow one to resize."' failed.
Aborted (core dumped)
```

christian-rauch, 2018-12-13:
Important bits from backtrace:
```
#4  0x00007fffcbb3bf6d in Eigen::DenseBase<Eigen::Ref<Eigen::Matrix<double, -1, -1, 0, -1, -1>, 0, Eigen::OuterStride<-1> > >::resize(long, long) ()
   from /tmp/track/devel/lib//libexotica_aico_solver.so
#5  0x00007fffca66dbcd in exotica::OMPLSolver<exotica::SamplingProblem>::GetPath(Eigen::Ref<Eigen::Matrix<double, -1, -1, 0, -1, -1>, 0, Eigen::OuterStride<-1> >, ompl::base::PlannerTerminationCondition&) ()
   from /tmp/track/devel/lib//libexotica_ompl_solver.so
```

It is related to https://github.com/ipab-slmc/exotica/blob/0fa07f062b8dbcf15807705ea59e793a2c966ad2/exotations/solvers/exotica_ompl_solver/src/ompl_solver.cpp#L193

wxmerkt, 2018-12-14:
Fixed in #435.

closed 2018-12-14 by wxmerkt