ipab-slmc/exotica #447: Published collision objects get deleted and added alternating

christian-rauch, 2018-12-17:
The collision scene object published via `publishPose(solution, ompl.getProblem())` alternative between `ADD` and `DELETEALL` causing flickering in RViz.
Instead of alternate publishing of the two actions, they should be added together in a single `MarkerArray` such that old markers are removed and new markers are added at the same time.

Note: If the IDs of marker objects do not change, there is no need to remove markers since old markers with the same ID can simpyl be overwritten.

VladimirIvan, 2018-12-17:
That would be good.
Do we have unique markers though?
The objects get destroyed internally if you reload the scene (I think).
Their names may be unique. I'm not sure if there is an easy way to get an ID for this.

wxmerkt, 2018-12-17:
@VladimirIvan Christian already provided a minimal fix, cf. #448

christian-rauch, 2018-12-17:
They have all the same namespace (`mrk.ns = "CollisionObjects";`) with enumerated IDs (`mrk.id = i;`).

The PR publishes all markers together with an optional `DELETEALL` action. I.e. the `DELETEALL` is added by `resetModel`. This covers the case where you can publish additional shapes to the `CollisionObjects` maker namespace. If we know that we always publish a complete collision scene, we could add the `DELETEALL` always when republishing a new set of collision shapes.

wxmerkt, 2018-12-17:
Your above proposed solution is what I used in a private/not yet merged change (it was a online change to `debugSceneChanged` and `publishFrames`. I prefer your new solution with reusing the marker array message though.

PR #448 (2018-12-18): Publish single list of markers with add and delete actions
This PR adds a global list of published markers to `KinematicTree`, including actions for `ADD` and `DELETEALL` (fixes #447 ).

- `resetModel` adds a `DELETEALL` marker to this global list, but does not publish them immediately
- the global list is published in `publishFrames` if the scene changed
- using enum `DELETEALL` (not available in indigo) for better readability and compatibiity with ROS message
closed 2018-12-18 by wxmerkt