ipab-slmc/exotica #452: strings returned by pyexotica are unicode

christian-rauch, 2018-12-17:
This might actually not be an issue, but a feature of `pybind11`: https://pybind11.readthedocs.io/en/stable/advanced/cast/strings.html

Strings that are returned from python wrapped functions, e.g. `problem.getScene().getRootFrameName()` or `problem.getScene().getJointNames()` are in unicode format.

This is problematic if you want to use these string to format ROS messages since the serialisation expects all strings to be ASCII format.

E.g.: using a string as frame_id in a message Header:
```Python
exo_root_frame = self.problem.getScene().getRootFrameName()
hdr = Header(stamp=rospy.get_rostime(), frame_id=exo_root_frame)
print(hdr)
```
will show like:
```
seq: 0
stamp: 
  secs: 1545070715
  nsecs: 596716880
frame_id: !!python/unicode "world"
```
and result in a serialisation error:
```
Traceback (most recent call last):
  File "./src/sample_main.py", line 12, in <module>
    Sample(config=conf).run()
  File "/home/christian/Development/mpe_ws/src/robot_state_sampling/src/sample.py", line 176, in run
    self.pub_state_goal.publish(state=rs)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/topics.py", line 886, in publish
    raise ROSSerializationException(str(e))
rospy.exceptions.ROSSerializationException: field state.joint_state.header.frame_id is a unicode string instead of an ascii string
```

I assume that this issue will not (and cannot) be fixed within EXOTica. This issue is merely a note.

wxmerkt, 2018-12-17:
It is a feature indeed rather than an issue. Simply add `.encode('utf-8')` after the method call to get the underlying string.

christian-rauch, 2018-12-17:
I believe we do not need unicode link and joint names. `PyUnicode_DecodeASCII` would decode the C++ strings to ASCII in the binding, but it's not a good solution to maintain that all strings in pyexotica are using this conversion function.
Maybe in the future, there will be a way to tell pybind globally to use ASCII rather than UTF-8 for the given module.

VladimirIvan, 2018-12-17:
This is handled [here](https://github.com/ipab-slmc/exotica/blob/master/exotica_python/src/Exotica.cpp#L8) for everything we wrote. This was working fine on 14.04. The behaviour might have changed with with a newer version of python/pybind11/ros. It won't convert output of functions that return std::string though, only arguments of the ones we defined.

Furthermore Python3 went completely away from ASCII, they use Unicode on all strings. ROS is still using Python2.7.

I'm in favor of Wolf's solution unless there is some pybind11 magic that can fix this.

christian-rauch, 2018-12-17:
Python 2 EOL is January 1st, 2020 (finally). So ROS1 needs to support Python 3 until then anyway, ROS2 does now already. So this issue should vanish by itself within 1 year and a couple of days.

wxmerkt, 2018-12-18:
Closing this, the solution with `.decode` should work. Please feel free to reopen if not resolved.

closed 2018-12-18 by wxmerkt