ipab-slmc/exotica #460: 'getSolver' does not return a solver

christian-rauch, 2018-12-18:
The method `Scene::getSolver()` is rather badly named since it actually returns a `KinematicTree` and not a solver as in `exotica::MotionSolver`.

This is especially confusing if used with the Python bindings, since after setting up a solver and a problem like:
```Python
self.solver = exo.Setup.createSolver(...)
self.problem = exo.Setup.createProblem(...)
self.solver.specifyProblem(self.problem)
```
a further call to:
```Python
problem_solver = self.problem.getScene().getSolver()
```
suggests that `problem_solver` is the same instance as the `self.solver` defined above.

The method `Scene::getSolver()` should be renamed to something like `Scene::getKinematicTree()`.

wxmerkt, 2018-12-18:
I agree. The source for this was that the `KinematicTree` is _one potential_ solver for the scene/planning scene - and should be both a kinematic and dynamic solver. Awaiting suggestions @VladimirIvan?

VladimirIvan, 2018-12-18:
Lets go with `Scene::getKinematicTree()`

wxmerkt, 2018-12-18:
Let's block this change until after #457 is merged.

christian-rauch, 2018-12-18:
Shall we delay this change until after the next release? It will break a lot of user code.

wxmerkt, 2018-12-18:
#457 and the upcoming changes to exotica core will break most user code. We had a discussion and it was decided to go do all the hard renaming now, release 5.0 as binary, then keep the API stable and all new code must adhere to the style guide.

PR #457 (2018-12-18): Rename/update API exotica_python
- Renames entire API in `exotica_python` (minor exception where it mirrors KDL C++ API)
- Updates all examples in `exotica_examples` to comply
- Fixes some examples that were broken

Relates to #421 

Fixes #460
closed 2018-12-18 by VladimirIvan