ipab-slmc/exotica #481: Examples scripts not running

cmower, 2019-01-08:
- my system is ubuntu 16 and i'm using ros kinetic, I have created a completely clean catkin workspace and added `exotica` as per [here](https://github.com/ipab-slmc/exotica#installation), I am on  latest master branch
- everything builds as expected, with no errors or warnings (apart from `[Ignored] exotica` that I am assuming after all the recent changes is to be expected)
```
[build] Successful packages:                                                                                                                                                                                                          
                                                                                                                                                                                                                                      
 [   Ignored] exotica                                 [Successful] exotica_aico_solver                     [Successful] exotica_collision_scene_fcl             [Successful] exotica_collision_scene_fcl_latest                       
 [Successful] exotica_core                            [Successful] exotica_core_task_maps                  [Successful] exotica_examples                        [Successful] exotica_ik_solver                                        
 [Successful] exotica_levenberg_marquardt_solver      [Successful] exotica_ompl_solver                     [Successful] exotica_python                          [Successful] exotica_time_indexed_rrt_connect_solver  
```
- I have tried launching the following 
  - `python_ik.launch`
  - `python_ik_eff_axis_alignment.launch`
  - `cpp_ik_minimal.launch`
  - `cpp_aico.launch`
- each fail, i.e. they seg-fault when `solver.solve()`/`Solve()` is called
- I ran `$ roslaunch exotica_examples python_ik.launch debug:=True` and the copied the output of the back trace below, seems something is going wrong with Eigen but I'm not sure what this means?
- I have tried updating/upgrading my system to no avail (i.e. `sudo apt-get update` and `sudo apt-get upgrade`)
- Seems only solution left is to format my laptop and start over, unless anyone has any better ideas?

```
GNU gdb (Ubuntu 7.11.1-0ubuntu1~16.5) 7.11.1
Copyright (C) 2016 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from python...(no debugging symbols found)...done.
(gdb) run
Starting program: /usr/bin/python /home/chris/Projects/exotica_dev_ws/src/exotica/exotica_examples/scripts/example_ik __name:=example_ik_node __log:=/home/chris/.ros/log/a968d426-1368-11e9-9cf0-40490ffece99/example_ik_node-2.log
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7fffe1294700 (LWP 16189)]
[New Thread 0x7fffdea93700 (LWP 16190)]
[New Thread 0x7fffdc292700 (LWP 16191)]
[New Thread 0x7fffd9a91700 (LWP 16192)]
[New Thread 0x7fffd7290700 (LWP 16193)]
[New Thread 0x7fffd6a8f700 (LWP 16194)]
[New Thread 0x7fffd428e700 (LWP 16195)]
[New Thread 0x7fffc5665700 (LWP 16197)]
[New Thread 0x7fffc4e64700 (LWP 16198)]
[New Thread 0x7fffbffff700 (LWP 16199)]
[New Thread 0x7fffbf7fe700 (LWP 16204)]
[New Thread 0x7fffbeffd700 (LWP 16206)]
[New Thread 0x7fffbe7fc700 (LWP 16207)]
[ INFO] [1546967599.776426472]: Loading robot model 'lwr'...
Publishing IK
python: /usr/local/include/eigen3/Eigen/src/QR/ColPivHouseholderQR.h:380: Eigen::ColPivHouseholderQR<MatrixType>::RealScalar Eigen::ColPivHouseholderQR<MatrixType>::threshold() const [with _MatrixType = Eigen::Matrix<double, -1, -1>; Eigen::ColPivHouseholderQR<MatrixType>::RealScalar = double]: Assertion `m_isInitialized || m_usePrescribedThreshold' failed.

Thread 1 "python" received signal SIGABRT, Aborted.
0x00007ffff7825428 in __GI_raise (sig=sig@entry=6)
    at ../sysdeps/unix/sysv/linux/raise.c:54
54      ../sysdeps/unix/sysv/linux/raise.c: No such file or directory.
(gdb) bt
#0  0x00007ffff7825428 in __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:54
#1  0x00007ffff782702a in __GI_abort () at abort.c:89
#2  0x00007ffff781dbd7 in __assert_fail_base (fmt=<optimised out>, assertion=assertion@entry=0x7fffcc084fd8 "m_isInitialized || m_usePrescribedThreshold", file=file@entry=0x7fffcc084df0 "/usr/local/include/eigen3/Eigen/src/QR/ColPivHouseholderQR.h", line=line@entry=380, 
    function=function@entry=0x7fffcc08e200 <Eigen::ColPivHouseholderQR<Eigen::Matrix<double, -1, -1, 0, -1, -1> >::threshold() const::__PRETTY_FUNCTION__> "Eigen::ColPivHouseholderQR<MatrixType>::RealScalar Eigen::ColPivHouseholderQR<MatrixType>::threshold() const [with _MatrixType = Eigen::Matrix<double, -1, -1>; Eigen::ColPivHouseholderQR<MatrixType>::"...) at assert.c:92
#3  0x00007ffff781dc82 in __GI___assert_fail (assertion=0x7fffcc084fd8 "m_isInitialized || m_usePrescribedThreshold", file=0x7fffcc084df0 "/usr/local/include/eigen3/Eigen/src/QR/ColPivHouseholderQR.h", line=380, 
    function=0x7fffcc08e200 <Eigen::ColPivHouseholderQR<Eigen::Matrix<double, -1, -1, 0, -1, -1> >::threshold() const::__PRETTY_FUNCTION__> "Eigen::ColPivHouseholderQR<MatrixType>::RealScalar Eigen::ColPivHouseholderQR<MatrixType>::threshold() const [with _MatrixType = Eigen::Matrix<double, -1, -1>; Eigen::ColPivHouseholderQR<MatrixType>::"...) at assert.c:101
#4  0x00007fffcc020bab in Eigen::ColPivHouseholderQR<Eigen::Matrix<double, -1, -1, 0, -1, -1> >::threshold (this=0x7fffffffc640) at /usr/local/include/eigen3/Eigen/src/QR/ColPivHouseholderQR.h:380
#5  0x00007fffcc01c887 in Eigen::ColPivHouseholderQR<Eigen::Matrix<double, -1, -1, 0, -1, -1> >::rank (this=0x7fffffffc640) at /usr/local/include/eigen3/Eigen/src/QR/ColPivHouseholderQR.h:259
#6  0x00007fffcc0190a7 in Eigen::CompleteOrthogonalDecomposition<Eigen::Matrix<double, -1, -1, 0, -1, -1> >::computeInPlace (this=0x7fffffffc640) at /usr/local/include/eigen3/Eigen/src/QR/CompleteOrthogonalDecomposition.h:417
#7  0x00007fffcc0161ab in Eigen::CompleteOrthogonalDecomposition<Eigen::Matrix<double, -1, -1, 0, -1, -1> >::compute<Eigen::Matrix<double, -1, -1, 0, -1, -1> > (this=0x7fffffffc640, matrix=...) at /usr/local/include/eigen3/Eigen/src/QR/CompleteOrthogonalDecomposition.h:187
#8  0x00007fffcc012b67 in Eigen::CompleteOrthogonalDecomposition<Eigen::Matrix<double, -1, -1, 0, -1, -1> >::CompleteOrthogonalDecomposition<Eigen::Matrix<double, -1, -1, 0, -1, -1> > (this=0x7fffffffc640, matrix=...) at /usr/local/include/eigen3/Eigen/src/QR/CompleteOrthogonalDecomposition.h:118
#9  0x00007fffcc00f15a in Eigen::MatrixBase<Eigen::Matrix<double, -1, -1, 0, -1, -1> >::completeOrthogonalDecomposition (this=0x7fffffffc5c0) at /usr/local/include/eigen3/Eigen/src/QR/CompleteOrthogonalDecomposition.h:557
#10 0x00007fffcc007435 in exotica::IKSolver::Solve (this=0x1790e10, solution=...) at /home/chris/Projects/exotica_dev_ws/src/exotica/exotations/solvers/exotica_ik_solver/src/ik_solver.cpp:125
#11 0x00007ffff673db02 in <lambda(std::shared_ptr<exotica::MotionSolver>)>::operator()(std::shared_ptr<exotica::MotionSolver>) const (__closure=0xadf058, sol=std::shared_ptr (count 3, weak 0) 0x1790e10) at /home/chris/Projects/exotica_dev_ws/src/exotica/exotica_python/src/pyexotica.cpp:691
#12 0x00007ffff6761396 in pybind11::detail::argument_loader<std::shared_ptr<exotica::MotionSolver> >::call_impl<Eigen::Matrix<double, -1, -1>, pybind11_init__pyexotica(pybind11::module&)::<lambda(std::shared_ptr<exotica::MotionSolver>)>&, 0ul, pybind11::detail::void_type>(<lambda(std::shared_ptr<exotica::MotionSolver>)> &, std::index_sequence, <unknown type in /home/chris/Projects/exotica_dev_ws/devel/lib/python2.7/dist-packages/_pyexotica.so, CU 0x0, DIE 0x3d8a91>) (this=0x7fffffffca40, f=...) at /opt/ros/kinetic/share/pybind11_catkin/cmake/../../../include/pybind11_catkin/pybind11/cast.h:1870
[... cut ...]
#17 0x00000000004c5326 in PyEval_EvalFrameEx ()
#18 0x00000000004b9b66 in PyEval_EvalCodeEx ()
#19 0x00000000004eb69f in ?? ()
#20 0x00000000004e58f2 in PyRun_FileExFlags ()
#21 0x00000000004e41a6 in PyRun_SimpleFileExFlags ()
#22 0x00000000004938ce in Py_Main ()
#23 0x00007ffff7810830 in __libc_start_main (main=0x493370 <main>, argc=4, argv=0x7fffffffd308, init=<optimised out>, fini=<optimised out>, rtld_fini=<optimised out>, stack_end=0x7fffffffd2f8) at ../csu/libc-start.c:291
#24 0x0000000000493299 in _start ()
(gdb) 

```

wxmerkt, 2019-01-08:
Do you build in Debug, Release, or with an optimization flag enabled?

cmower, 2019-01-08:
I always use`$ catkin build -s` , nothing more fancy than that.

wxmerkt, 2019-01-08:
I just setup a fresh exotica copy on the Husky workstation (16.04) which works fine (no additional cmake args, i.e., probably the same setup you have. My 18.04 setups on both laptop and computer work fine as well. 
The error in `ColPivHousehoulderQR` (not initialised) _seems_ to point to an Eigen issue on your machine: e.g., a compilation issue or a differing version. You could remove `build`, `devel`, `logs`, `.catkin_tools` (then reinitialise with `catkin config --init`), but it may reoccur. You could also compile with explicit debug: `catkin config --cmake-args -DCMAKE_BUILD_TYPE=Debug`. But for now and for us this is irreproducible.

VladimirIvan, 2019-01-08:
As of the last PR, all examples run on 18.04 out of the box. I haven't checked 16.04 but I did check ColPivHouseholderQR on 18.04 before committing it. There were no issues.

cmower, 2019-01-09:
I have tried
* reinstalling `libeigen3-dev` and  `ros-kinetic-desktop`
* removing `build`, `devel`, `logs`, `.catkin_tools`, re-initializing with `catkin config --init`, and compiling with debug (i.e. `catkin config --cmake-args -DCMAKE_BUILD_TYPE=Debug`)

No luck! Will start prepping for and carry out system format and re-install with Ubuntu 18 and ROS melodic.

closed 2019-01-09 by wxmerkt