ipab-slmc/exotica #483: Use of rotation costs in EffOrientation

christian-rauch, 2019-01-08:
The `EffOrientation` task map converts any provided rotation format (Quaternion, Euler angles, ...) into a variable length task space vector (e.g Quaternion -> 4D, RPY -> 3D, Rotation matrix -> 9D, axis-angle -> 3D) but uses the rotation part of the twist (3D) of the Jacobian as gradient for all of these task vectors.

This doesn't seem appropriate. In fact, this is only working if the RPY rotation type is used since the rotation of the twist is represented as RPY.
If the rotation part of the twist is used as gradient, the task vector needs to be constructed from the corresponding 3 Euler angles.

As alternative, the magnitude of a rotation is usually the absolute angle of the axis-angle representation. However, the gradients for this cannot directly be obtained from the Jacobian.

christian-rauch, 2019-01-08:
I think we should simply convert the `KDL::Frame::Rotation` (3x3 matrix) to RPY using `GetRPY()` and set `TaskSpaceDim` to 3 per frame.
Any thoughts?

VladimirIvan, 2019-01-08:
I don't understand what is the problem?

Any orientation in the TaskSpaceVector is stored as Quaternion, Euler angles, ...
The minus operator converts it into a full KDL::Rotation (matrix), then computes the SO3 difference and returns the angular twist.

Changing between the different representations has no effect on the result apart from accuracy loss in the conversion into a full matrix.

The changes in the TaskSpaceVector in #484 will remove this. This means that every subtraction will be treated as a vector subtraction ([RPY]-[RPY]). This will result in an incorrect (non-geometric) angle difference and it will suffer from gimbal lock.

>> As alternative, the magnitude of a rotation is usually the absolute angle of the axis-angle representation. However, the gradients for this cannot directly be obtained from the Jacobian.

What do you mean by this? You can choose angle-axis representation and still get a Jacobian.

christian-rauch, 2019-01-09:
I overlooked that the `TaskSpaceVector` has its own subtraction operator.
The `TaskSpaceVector` is storing any length of untyped raw vector (e.g. 4D for quaternion) but its subtraction operator (optionally) converts sections of this raw vector into a `KDL::Rotation`, using the rotation types provided in a list of `TaskVectorEntry` that can be defined by method `GetLieGroupIndices`. The returned `rotvec` is in angle axis representation, i.e. the axis with length of the rotation in rad.
(Is this correct?)

I see that if I remove the rotation type (`GetLieGroupIndices`) no `TaskVectorEntry` will be added and the subtraction operator will use the "normal" element-wise subtraction of the raw vector data. I.e. the RPY vector will not be seen as a rotation.

However, I still do not see why the `EffOrientation` task map needs the `Type` parameter. In `EffOrientation` the `rotation_type_` is used to convert a `KDL::Rotation` into a serialised vector for Phi. During the subtraction operation, this is converted back. I.e. the `rotation_type_` is just an intermediate representation for serialisation.
Why does this need to be configurable? Wouldn't it make sense to simply select the quaternion or rotation matrix representation? If there is an issue with the Gimbal lock, why is RPY selected as the default rotation representation?

VladimirIvan, 2019-01-10:
The minus operator will eliminate the gimbal lock (even for RPY) because the difference is done by the operation in SO3. So no matter what `rotation_type_` you choose, the resulting `y_diff` will be same.

The `rotation_type_` only decides how will the task space be stored (3, 4 or 12 vector for Euler, quaternion, matrix respectively). This affects the format in which you have to provide the goal (`SetGoal`). It's not required beyond that as long as you choose a default type (e.g., RPY). Quaternion and especially the matrix use more space to store the same rotation, although this is negligible in grand scheme of things. I haven't event seen any benefit in using the matrix for speed (no conversion needed).

I have originally started with only RPY, but then I realized we could use any SO3 parametrisation so I added the other ones as an exercise and for convenience when setting the goal.

christian-rauch, 2019-01-10:
Since `PlanningProblem::SetGoal` is directly writing into the task space vector, it makes sense to have a user-defined rotation representation. I actually never used `SetGoal` and therefore never encountered this issue. I.e. I always re-instantiated the task map where the conversion to `KDL::Frame` is done automatically.

christian-rauch, 2019-01-10:
`SetGoal` could actually convert rotations automatically as it is done in the `GetFrame`:
https://github.com/ipab-slmc/exotica/blob/cad9fbad9906cefed16cf0d16fbe05dc9c163a23/exotica_core/src/tools/conversions.cpp#L51-L68
E.g. the user would need to tell `SetGoal` with an additional parameter (type: raw vector / pose / rotation) that the provided vector is a full pose or rotation and this would be converted into a `KDL::Frame` or `KDL::Rotation` and then into a common EXOTica-wide rotation representation. This would remove the need to store the information how orientations are represented internally, the different conversion functions and the parameter for the task maps.

VladimirIvan, 2019-01-10:
If we went with one of the 3D representations (eg RPY), we could also remove the concept of task size and jacobian size.

However, SetGoal is a member variable of specific planning problems. The conversion code would have to be either implemented in the problem, or even better, in the task. A lot of repeated overloading may be required. Let me know if you come up with an elegant solution.

closed 2020-11-04 by wxmerkt