ipab-slmc/exotica #494: Use PyKDL instead of own pybind11 definitions

christian-rauch, 2019-01-16:
`pyexotica` defines python bindings for some of the KDL types (`KDL::Frame`, `KDL::Vector`, `KDL::RotationalInertia`, `KDL::RigidBodyInertia`).
I think it would make sense to replace them by the "official" python types in [`PyKDL`](https://github.com/orocos/orocos_kinematics_dynamics/tree/master/python_orocos_kdl).

Is it possible to reuse python bindings from another wrapper layer (in this case [SIP](https://wiki.python.org/moin/SIP)) in pybind11 for EXOTica?

wxmerkt, 2019-01-16:
I don't think this works or is a good idea. It would need to match the correct C++ classes after translation. It would likely be a lot of work if it works at all. Do you see an immediate need?

christian-rauch, 2019-01-16:
No, it's not urgent. It's just that we are duplicating work that has been done in the upstream KDL project.
It would also be advantageous, if exotica could interface this way with other python libraries that use the upstream bindings, e.g. the conversion function in rospy.

By:
> It would need to match the correct C++ classes after translation.

do you mean the C++ classes of the wrapped bindings? E.g. that `py::class_<KDL::Frame>` matches with the `Frame` defined in upstream's [`frames.sip`](https://github.com/orocos/orocos_kinematics_dynamics/blob/9ff3052d6b2799bc039bccdab8f85b61367026eb/python_orocos_kdl/PyKDL/frames.sip#L198)?

wxmerkt, 2019-01-16:
I think it's quite an effort to validate this or get it to work - if it works at all (without being too big of a hack). And given that the benefit is minor, for now (conversions are four to seven lines...), with no extra cycles to spare I do not see us implementing it.

The C++ side needs to be the native KDL::Frame as it's being passed to the methods.