ipab-slmc/exotica #504: Independent scene and problem construction

christian-rauch, 2019-01-29:
At the moment, `Setup.create_problem` creates the kinematic tree / scene and the problem (incl. task maps) at the same time. I would like to add frames to the scene (via `AddElement` or `AddEnvironmentElement`) that are also used in task maps. This is currently not possible.
You can create the problem without task maps to access the scene, but this will be overwritten when recreating the problem with actual task maps.

A concrete example would be a camera frame, that is not part of the kinematic chain, but which is used in a EffFrame task map to reach a certain end-effector pose given a static transform (not controlled) or another constraint via a second EffFrame task map.

I think there should be a `Setup.create_scene` function to provide a scene handle for the user and a `Setup.create_problem()` that takes the problem type, the scene and the task maps with their rho. This way, a user can manipulate the scene and define task maps that relate to added scene frames.

wxmerkt, 2019-01-29:
You can add custom links to the `Scene` initializer - does this solve your use case? Have a look at the attach demo if you're looking for an example. You can then `attach_object_local` to set the actual transformation of that initialized link.

VladimirIvan, 2019-01-29:
Wolf just pointed you to the wrong example.
Have look at [example_aico_trajectory](https://github.com/ipab-slmc/exotica/blob/a9e3f2406ce3102c468a205b72c835c44bc18e3f/exotica_examples/resources/configs/example_aico_trajectory.xml#L16).
It adds links to the initializer. These get created before the the TaskMaps so you get to use them there.
These links will be named so you can attach them as you please at any point. Especially if you want to reposition them at runtime.

You can also load the initializer, edit it (add frames) and then create the problem. We had this type of initialization in the examples but it was confusing for new users so now it only exists in tests. Have a look at [this one](https://github.com/ipab-slmc/exotica/blob/a79a99024c83cbbfdf281c877e685bebdee2de7e/exotica_examples/test/test_valkyrie_collision_check_fcl_default#L15). We use the initializer editing heavily in the [task map test](https://github.com/ipab-slmc/exotica/blob/a79a99024c83cbbfdf281c877e685bebdee2de7e/exotations/exotica_core_task_maps/test/test_maps.cpp#L284).

christian-rauch, 2019-01-30:
Thanks Vlad, I didn't think about using the initialiser before creating the problem.

To add a `camera_frame` frame that is used by task maps, I directly add the frames to the scene initialiser:
```Python
scene = {'URDF': urdf_path, 'SRDF': srdf_path, 'JointGroup': self.conf['joint_group']}
# add additional frame
scene.update({'Links': [('exotica/Link', {'Name': "camera_frame", 'Transform': [0,0,0,1,0,0,0]})]})
pose_frame = [('exotica/Frame', {'Link': "camera_frame", 'Base': ""})]
self.problem = exo.Setup.create_problem(
    ('exotica/UnconstrainedEndPoseProblem',
        {'Name': 'problem',
        'PlanningScene': [('exotica/Scene', scene)],
        # add separately weighted position and orientation tasks
        'Maps': [
            ('exotica/EffPosition', {'Name': "position", 'EndEffector': pose_frame}),
            ('exotica/EffOrientation', {'Name': "orientation", 'EndEffector': pose_frame, 'Type': 'Quaternion'}),
            ('exotica/JointLimit', {'Name': "joint_limits", 'SafePercentage': str(0)}),
        ],
        'Cost': [
            ('exotica/Task', {'Task': "position", 'Rho': str(10)}),
            ('exotica/Task', {'Task': "orientation", 'Rho': str(1)}),
            ('exotica/Task', {'Task': "joint_limits", 'Rho': str(1)}),
        ]
        }))
```

It seems to work because the initiliser doesn't fail.

A minor issue is however that I don't know in advance, which links/frames are already in the scene. If a scene handle would be available, I could test against `get_model_link_names()` which frames are already present.

VladimirIvan, 2019-01-30:
`KinematicTree::GetTreeMap` contains all the defined frames and their names. You can expose the frame names by collecting all the keys of that map. This could sit in the `Scene` for convenience.

closed 2019-02-18 by christian-rauch