ipab-slmc/exotica #513: Convention for optimisation solvers

wxmerkt, 2019-02-13:
Considering a trajectory consisting out of `T` states some of our optimisation solvers only consider `[1..T]` (e.g., SNOPT) while others optimise the full trajectory `[0..T]`. AICO considers the full trajectory `[0..T]` but only updates `[1..T]` during the sweeps.

There are a few arguments for and against:

## For [1..T]
- Smaller optimisation problems (smaller by `N`)
- The start state can be set using `.start_state` - as is available e.g. on robots from sensing.
- Solvers following this: SNOPT, OPTPP

## For [0..T]
- Different treatment of the first timestep makes indexing harder and is error prone.
- In planning problems alone, we do not always have a start state (unless running IK separately - often as a separate problem/step).
- We can specify constraints along the entire trajectory, including the first timestep.
- Easier to specify initial trajectory - in the other case we need a mix between start state and initial trajectory (where first state of initial trajectory => start state)
- Solvers following this: IPOPT, L-BFGS, SciPy

I personally think `[0..T]` would allow easier implementation of solvers. The problems will be slightly larger but it would not require having to have the extra logic and the need for separate IK for planning the start state.

What do you think @VladimirIvan?

VladimirIvan, 2019-02-13:
So do you propose to: 
 - (a) Use state `[0]` as a control variable?
 - (b) Not use `[0]` as a control variable in the solver but leave the option to specify Rho/Goal/... ?
Where would this be implemented? Does it require changes in the problem?

wxmerkt, 2019-02-18:
I went for changing to [1..T] in the problem and trying to get all solvers to directly use problem getters for equality/inequality constraints. This way indexing only has to be figured out once and writing solvers will get easier in the future.

PR #520 (2019-02-25): TimeIndexedProblem: Getters for sparse Jacobians; move to "only active constraints"
This PR contains many updates that improve the efficiency of constrained trajectory problems - e.g., with SNOPT >20%. Further it simplifies the implementation of constrained solvers and avoids mistakes in indexing for the sparse Jacobian matrices. This is accomplished by:

- Adding methods to get the triplets (row, column, value) of the sparse Jacobians for equality and inequality constraints. This avoids having to redo indexing per solver (which is error-prone). It is also faster and adds readability - in this [... cut]
closed 2019-02-25 by wxmerkt