ipab-slmc/exotica #524: Memory issues in repeated optimisation of problems involving EffFrame

wxmerkt, 2019-03-01:
This may relate to our issues involving optimisation of frames/orientations:

- When running multiple, supposedly independent runs we get memory corruption. I've been noticing this in both Python and C++ and it can either slow repeated optimisations down or make the results "arbitrary" when using deterministic solvers.

VladimirIvan, 2019-03-01:
Can you provide code to reproduce this?

wxmerkt, 2019-03-01:
Yes, after the deadline. I noticed it does not happen for the LWR, Franka, and Ada examples but for the Quadrotor (which is floating-base only - perhaps that's the source?).

wxmerkt, 2019-03-27:
I have been debugging this and it *may* be related to SNOPT and constrained problems:

- Persists across both Python and C++
- Can be reproduced with and without EffFrame/any taskmap (unlike initial report).
- Not reproducible with SNOPT on unconstrained trajectory problems.

wxmerkt, 2019-03-27:
Tracked down to a off-by-1 bug in the _constrained_ SNOPT trajectory optimization. Also harmonised constrained/unconstrained SNOPT trajectory solvers in the process.

closed 2019-03-27 by wxmerkt