ipab-slmc/exotica #527: get_controlled_base_type always returns BaseType.Fixed

christian-rauch, 2019-03-11:
If my SRDF defines a virtual `fixed` joint, the code:
```Python
base_type = self.problem.get_scene().get_kinematic_tree().get_controlled_base_type()
print("base type:", base_type)
print("model joints", self.problem.get_scene().get_model_joint_names())
```
correctly returns:
```
base type: BaseType.Fixed
model joints [u'lwr_arm_0_joint', u'lwr_arm_1_joint', u'lwr_arm_2_joint', u'lwr_arm_3_joint', ...]
```
i.e. a fixed base with no base joints.

But if I chane the virtual joint to `floating` or `planar`, the same code returns:
```
base type: BaseType.Fixed
model joints [u'table_lwr_arm_base_joint/trans_x', u'table_lwr_arm_base_joint/trans_y', u'table_lwr_arm_base_joint/trans_z', u'table_lwr_arm_base_joint/rot_x', u'table_lwr_arm_base_joint/rot_y', u'table_lwr_arm_base_joint/rot_z', u'lwr_arm_0_joint', u'lwr_arm_1_joint', u'lwr_arm_2_joint', u'lwr_arm_3_joint', ...]
```
```
base type: BaseType.Fixed
model joints [u'table_lwr_arm_base_joint/x', u'table_lwr_arm_base_joint/y', u'table_lwr_arm_base_joint/theta', u'lwr_arm_0_joint', u'lwr_arm_1_joint', u'lwr_arm_2_joint', u'lwr_arm_3_joint', ...]
```
i.e. the correct base joints but it assumes a fixed base.

wxmerkt, 2019-03-11:
Is the virtual joint part of your planning group? Can you send us some files to reproduce?

christian-rauch, 2019-03-11:
This happens for example with the `exotica/exotica_examples/scripts/example_ik` and `exotica/exotica_examples/resources/robots/lwr_simplified.srdf`.

If you change the the virtual joint to `floating` and print the base type according to this diff:
```diff
diff --git a/exotica_examples/resources/robots/lwr_simplified.srdf b/exotica_examples/resources/robots/lwr_simplified.srdf
index 633a67df..dce01854 100644
--- a/exotica_examples/resources/robots/lwr_simplified.srdf
+++ b/exotica_examples/resources/robots/lwr_simplified.srdf
@@ -15,7 +15,7 @@
     <!--END EFFECTOR: Purpose: Represent information about an end effector.-->
     <!--<end_effector name="eff" parent_link="lwr_arm_7_link" group="arm" />-->
     <!--VIRTUAL JOINT: Purpose: this element defines a virtual joint between a robot link and an external frame of reference (considered fixed with respect to the robot)-->
-    <virtual_joint name="world_joint" type="fixed" parent_frame="world_frame" child_link="base" />
+    <virtual_joint name="world_joint" type="floating" parent_frame="world_frame" child_link="base" />
     <!--DISABLE COLLISIONS: By default it is assumed that any link of the robot could potentially come into collision with any other link in the robot. This tag disables collision checking between a specified pair of links. -->
     <disable_collisions link1="base" link2="lwr_arm_0_link" reason="Adjacent" />
     <disable_collisions link1="base" link2="lwr_arm_1_link" reason="Never" />
diff --git a/exotica_examples/scripts/example_ik b/exotica_examples/scripts/example_ik
index 03ba7ba0..ee2e52b2 100755
--- a/exotica_examples/scripts/example_ik
+++ b/exotica_examples/scripts/example_ik
@@ -18,6 +18,8 @@ solver = exo.Setup.load_solver(
     '{exotica_examples}/resources/configs/example_ik.xml')
 problem = solver.get_problem()
 
+print("base type:", problem.get_scene().get_kinematic_tree().get_controlled_base_type())
+
 dt = 0.002
 t = 0.0
 q = array([0.0] * 7)
```

then the example programm wil print `BaseType.Fixed`.

wxmerkt, 2019-03-11:
Okay, you need an additional change in the SRDF for the world_joint to be part of the _planning group_: where the group "arm" is defined, you need to include "world_joint" (or change the planning group in your XML.

wxmerkt, 2019-03-11:
Have a look at this example:
https://github.com/ipab-slmc/exotica/blob/cebcde51a6bf89b57ab6b09e9f0f6c1061ce2e48/exotica_examples/resources/configs/example_ompl_freebase.xml#L11-L13

christian-rauch, 2019-03-11:
It makes sense that the virtual joint needs to be part of the planning group. It is however confusing that the base type is still fixed, although the model contains the joints.
Does this mean that a floating base, that is not part of the planning group, cannot be controlled but is still part of the robot state (a.k.a. underactuated)?

wxmerkt, 2019-03-11:
This is exactly the difference between `ModelBaseType` (which will be floating if the virtual joint is floating) and `ControlledBaseType` (which depends on whether the virtual joint is part of the planning group or not).

wxmerkt, 2019-03-12:
Can this be closed now?

christian-rauch, 2019-03-12:
Yes, I meant to close this with my last PR.

closed 2019-03-12 by christian-rauch