ipab-slmc/exotica #534: Cannot load virtual joint from standard SRDF

christian-rauch, 2019-03-18:
I am trying to load the SRDF of a Jaco robot (type `j2n6s300`) which is maintained by Kinova: [`j2n6s300.srdf`](https://github.com/Kinovarobotics/kinova-ros/blob/master/kinova_moveit/robot_configs/j2n6s300_moveit_config/config/j2n6s300.srdf). The URDF ([j2n6s300.urdf.txt](https://github.com/ipab-slmc/exotica/files/2979732/j2n6s300.urdf.txt)) is generated from the [`j2n6s300_standalone.xacro`](https://github.com/Kinovarobotics/kinova-ros/blob/master/kinova_description/urdf/j2n6s300_standalone.xacro).

Loading this SRDF generates the warning:
```
Skipping virtual joint 'world_joint' because its child frame 'root' does not match the URDF frame 'world'
No root/virtual joint specified in SRDF. Assuming fixed joint
```
and a call to:
```Python
scene = {'URDF': urdf_path, 'SRDF': srdf_path, 'JointGroup': "arm"}
model_links = exo.Setup.create_scene(('exotica/Scene', scene)).get_model_link_names()
```
fails with:
```
RuntimeError: In [...]/exotica/exotica_core/src/kinematic_tree.cpp
BuildTree 147
Can't initialize root joint!
```

It's not clear to me, why the virtual joint (`world_joint`) cannot be loaded. The same SRDF works fine in the MoveIt! demo.

christian-rauch, 2019-03-18:
I can load the SRDF if I remove link `world` and joint `connect_root_and_world` from the URDF, i.e. if I delete:
```XML
  <!-- for gazebo -->
  <link name="world"/>
  <joint name="connect_root_and_world" type="fixed">
    <child link="root"/>
    <parent link="world"/>
    <origin rpy="0 0 0" xyz="0 0 0"/>
  </joint>
```
In this case, the URDF joint `connect_root_and_world` and the virtual SRDF joint `world_joint` have the same parent and child link/frame. Is this something that should be supported by EXOTica?

wxmerkt, 2019-03-19:
I think there are two issues here:

- The Kinova URDF has a world link, which should not have been included in the first place.
- The SRDF uses the root link as the frame to which the virtual joint connects (which is correct, even though the definition seems to be fine with any frame on the robot, although I've only ever seen `base_link`/root of the tree).

And in combination with Exotica:

- Here, the virtual joint should in theory start at the base of the tree, not somewhere within it. The root of the tree in the URDF is `world`, not `root`. The relevant code is here:
https://github.com/ipab-slmc/exotica/blob/3305d49ee8a67953afb4bdf14747cef5083d4cf9/exotica_core/src/kinematic_tree.cpp#L136-L147

Most URDFs are fine and do not include the unnecessary world link (which technically shouldn't be included as it's what `virtual_joint` is for). If you find other examples, please let me know.

We are also happy to accept a pull request that also loads the kinematic chain prior to the virtual joint topologically (but accompanied with unit tests since this changes quite a bit!).

christian-rauch, 2019-03-19:
I agree that the `world` link should not be defined in the URDF. I removed it and sent a PR for this.

This issue has also been reported for the UR robots https://github.com/ros-industrial/universal_robot/issues/347 (with the same message that I got) and discussed in https://github.com/ros-industrial/universal_robot/pull/284.

The `world` frame is required by Gazebo: https://github.com/jacknlliu/development-issues/issues/68#issue-387792498, https://github.com/ros-industrial/universal_robot/pull/284#issuecomment-270779390

Given that the warning (`Skipping virtual joint ...`) is produced by MoveIt! I am not sure if we should implement a workaround in EXOTica or assume that the URDF+SRDF is compatible with MoveIt!.

christian-rauch, 2019-03-19:
As an alternative, is it possible to programmatically define the SRDF in EXOTica? E.g. not loading a SRDF file, but setting the `virtual_joint` and the `group` manually in C++ or Python?

VladimirIvan, 2019-03-20:
We use the MoveIt! robot model which requires both SRDF and URDF when loaded. There is no easy way to get rid of SRDF and I don't think we should. It defines all the additional information that is missing in the URDF.

By convention, every robot should have a root and that root should be connected to the world by a virtual joint (fixed, planar or floating). Exotica is relying on this convention. There is no workaround.

christian-rauch, 2019-03-20:
Alright. I agree that we should not create a workaround in EXOTica, but rely on workaround/fixes for URDF/SRDF files like mentioned in the references.

closed 2019-03-20 by christian-rauch