ipab-slmc/exotica #547: Unstable Jenkins builds

christian-rauch, 2019-04-29:
The build log (latest: http://build.ros.org/job/Mdev__exotica__ubuntu_bionic_amd64/33/consoleFull) contains:

> ```
> 18:34:38 [WARNINGS] Plug-in Result: Unstable - <a href="warningsResult">20 warnings</a> exceed the threshold of 0 by 20
> 18:34:38 Build step 'Scan for compiler warnings' changed build result to UNSTABLE
> ```

These 20 compiler warnings are caused by deprecated throw specifications (`warning: dynamic exception specifications are deprecated in C++11 [-Wdeprecated]`) in the `pinocchio` library:
```C++
17:53:23 In file included from /tmp/ws/src/exotica/exotations/dynamics_solvers/exotica_pinocchio_dynamics_solver/include/exotica_pinocchio_dynamics_solver/pinocchio_dynamics_solver.h:39:0,
17:53:23                  from /tmp/ws/src/exotica/exotations/dynamics_solvers/exotica_pinocchio_dynamics_solver/src/pinocchio_dynamics_solver.cpp:30:
17:53:23 /opt/ros/melodic/include/pinocchio/parsers/urdf.hpp:33:44: warning: dynamic exception specifications are deprecated in C++11 [-Wdeprecated]
17:53:23                 const bool verbose = false) throw (std::invalid_argument);
17:53:23                                             ^~~~~
17:53:23 /opt/ros/melodic/include/pinocchio/parsers/urdf.hpp:48:44: warning: dynamic exception specifications are deprecated in C++11 [-Wdeprecated]
17:53:23                 const bool verbose = false) throw (std::invalid_argument);
17:53:23                                             ^~~~~
17:53:23 /opt/ros/melodic/include/pinocchio/parsers/urdf.hpp:103:5: warning: dynamic exception specifications are deprecated in C++11 [-Wdeprecated]
17:53:23      throw (std::invalid_argument);
17:53:23      ^~~~~
17:53:23 /opt/ros/melodic/include/pinocchio/parsers/urdf.hpp:120:5: warning: dynamic exception specifications are deprecated in C++11 [-Wdeprecated]
17:53:23      throw (std::invalid_argument);
17:53:23      ^~~~~
17:53:23 /opt/ros/melodic/include/pinocchio/parsers/urdf.hpp:149:5: warning: dynamic exception specifications are deprecated in C++11 [-Wdeprecated]
17:53:23      throw (std::invalid_argument);
17:53:23      ^~~~~
17:53:23 /opt/ros/melodic/include/pinocchio/parsers/urdf.hpp:176:5: warning: dynamic exception specifications are deprecated in C++11 [-Wdeprecated]
17:53:23      throw (std::invalid_argument)
17:53:23      ^~~~~
17:53:23 /opt/ros/melodic/include/pinocchio/parsers/urdf.hpp:208:5: warning: dynamic exception specifications are deprecated in C++11 [-Wdeprecated]
17:53:23      throw (std::invalid_argument);
17:53:23      ^~~~~
17:53:23 /opt/ros/melodic/include/pinocchio/parsers/urdf.hpp:235:5: warning: dynamic exception specifications are deprecated in C++11 [-Wdeprecated]
17:53:23      throw (std::invalid_argument)
17:53:23      ^~~~~
17:53:23 In file included from /opt/ros/melodic/include/pinocchio/parsers/urdf.hpp:245:0,
17:53:23                  from /tmp/ws/src/exotica/exotations/dynamics_solvers/exotica_pinocchio_dynamics_solver/include/exotica_pinocchio_dynamics_solver/pinocchio_dynamics_solver.h:39,
17:53:23                  from /tmp/ws/src/exotica/exotations/dynamics_solvers/exotica_pinocchio_dynamics_solver/src/pinocchio_dynamics_solver.cpp:30:
17:53:23 /opt/ros/melodic/include/pinocchio/parsers/urdf/model.hxx:194:36: warning: dynamic exception specifications are deprecated in C++11 [-Wdeprecated]
17:53:23                       bool verbose) throw (std::invalid_argument)
17:53:23                                     ^~~~~
17:53:23 /opt/ros/melodic/include/pinocchio/parsers/urdf/model.hxx:539:46: warning: dynamic exception specifications are deprecated in C++11 [-Wdeprecated]
17:53:23                           const bool verbose) throw (std::invalid_argument)
17:53:23                                               ^~~~~
17:53:23 /opt/ros/melodic/include/pinocchio/parsers/urdf/model.hxx:563:46: warning: dynamic exception specifications are deprecated in C++11 [-Wdeprecated]
17:53:23                           const bool verbose) throw (std::invalid_argument)
17:53:23                                               ^~~~~
17:53:23 /opt/ros/melodic/include/pinocchio/parsers/urdf/model.hxx:616:7: warning: dynamic exception specifications are deprecated in C++11 [-Wdeprecated]
17:53:23        throw (std::invalid_argument)
17:53:23        ^~~~~
17:53:23 /opt/ros/melodic/include/pinocchio/parsers/urdf/model.hxx:636:5: warning: dynamic exception specifications are deprecated in C++11 [-Wdeprecated]
17:53:23      throw (std::invalid_argument)
17:53:23      ^~~~~
17:53:23 /opt/ros/melodic/include/pinocchio/parsers/urdf/model.hxx:657:43: warning: dynamic exception specifications are deprecated in C++11 [-Wdeprecated]
17:53:23                        const bool verbose) throw (std::invalid_argument)
17:53:23                                            ^~~~~
17:53:23 /opt/ros/melodic/include/pinocchio/parsers/urdf/model.hxx:676:43: warning: dynamic exception specifications are deprecated in C++11 [-Wdeprecated]
17:53:23                        const bool verbose) throw (std::invalid_argument)
17:53:23                                            ^~~~~
17:53:23 In file included from /opt/ros/melodic/include/pinocchio/parsers/urdf/geometry.hxx:10:0,
17:53:23                  from /opt/ros/melodic/include/pinocchio/parsers/urdf.hpp:246,
17:53:23                  from /tmp/ws/src/exotica/exotations/dynamics_solvers/exotica_pinocchio_dynamics_solver/include/exotica_pinocchio_dynamics_solver/pinocchio_dynamics_solver.h:39,
17:53:23                  from /tmp/ws/src/exotica/exotations/dynamics_solvers/exotica_pinocchio_dynamics_solver/src/pinocchio_dynamics_solver.cpp:30:
17:53:23 /opt/ros/melodic/include/pinocchio/parsers/utils.hpp:65:91: warning: dynamic exception specifications are deprecated in C++11 [-Wdeprecated]
17:53:23                                             const std::vector<std::string> & package_dirs) throw (std::invalid_argument)
17:53:23                                                                                            ^~~~~
17:53:24 In file included from /opt/ros/melodic/include/pinocchio/parsers/urdf.hpp:246:0,
17:53:24                  from /tmp/ws/src/exotica/exotations/dynamics_solvers/exotica_pinocchio_dynamics_solver/include/exotica_pinocchio_dynamics_solver/pinocchio_dynamics_solver.h:39,
17:53:24                  from /tmp/ws/src/exotica/exotations/dynamics_solvers/exotica_pinocchio_dynamics_solver/src/pinocchio_dynamics_solver.cpp:30:
17:53:24 /opt/ros/melodic/include/pinocchio/parsers/urdf/geometry.hxx:301:93: warning: dynamic exception specifications are deprecated in C++11 [-Wdeprecated]
17:53:24                                               const std::vector<std::string> & package_dirs) throw (std::invalid_argument)
17:53:24                                                                                              ^~~~~
17:53:24 /opt/ros/melodic/include/pinocchio/parsers/urdf/geometry.hxx:382:54: warning: dynamic exception specifications are deprecated in C++11 [-Wdeprecated]
17:53:24                              const GeometryType type) throw (std::invalid_argument)
17:53:24                                                       ^~~~~
17:53:24 In file included from /opt/ros/melodic/include/pinocchio/parsers/urdf.hpp:246:0,
17:53:24                  from /tmp/ws/src/exotica/exotations/dynamics_solvers/exotica_pinocchio_dynamics_solver/include/exotica_pinocchio_dynamics_solver/pinocchio_dynamics_solver.h:39,
17:53:24                  from /tmp/ws/src/exotica/exotations/dynamics_solvers/exotica_pinocchio_dynamics_solver/src/pinocchio_dynamics_solver.cpp:30:
17:53:24 /opt/ros/melodic/include/pinocchio/parsers/urdf/geometry.hxx:413:7: warning: dynamic exception specifications are deprecated in C++11 [-Wdeprecated]
17:53:24        throw(std::invalid_argument)
17:53:24        ^~~~~
17:53:24 /opt/ros/melodic/include/pinocchio/parsers/urdf/geometry.hxx:430:7: warning: dynamic exception specifications are deprecated in C++11 [-Wdeprecated]
17:53:24        throw(std::invalid_argument)
17:53:24        ^~~~~
```

To resolve these warnings, these throw specifiers need to be removed upstream via a PR or they need to be removed via downstream patches during the bloom release.

christian-rauch, 2019-04-29:
PR pending: https://github.com/stack-of-tasks/pinocchio/pull/788

wxmerkt, 2019-04-29:
Thank you for upstreaming the fix! We also have one "unstable" warning in the codebase between kinetic/melodic which will go away when we make another binary release.

christian-rauch, 2019-04-30:
The PR has been merged. Can we do a bloom release for `pinocchio_catkin`? The EXOTica Jenkins job should then pick up the new package at the next build.

wxmerkt, 2019-04-30:
Great! Ideally there would be a new minor release tag applied upstream so we can mirror that for `pinocchio_catkin`. We had also planned a change to how we wrap the library, but it may have to wait due to pecularities in the python-layout from upstream jrl-umi cmake (requiring `USE_DEB_LAYOUT` to be set manually).

wxmerkt, 2019-04-30:
Bloom releases are out:
- eigenpy v1.5.0 => v1.5.1
	- Kinetic: https://github.com/ros/rosdistro/pull/21063
	- Melodic: https://github.com/ros/rosdistro/pull/21062
- pinocchio v2.0.0 => v2.1.3
    - Kinetic: https://github.com/ros/rosdistro/pull/21066
    - Melodic: https://github.com/ros/rosdistro/pull/21065

I don't think the unstable build will fully go away though - I am still seeing dozens of warnings (`-Wdeprecated-declarations -Wunitialized` etc.). And part of it is our doing: the released version of exotica triggered a warning for the LM solver on Kinetic, but not on Melodic so was "unstable" between versions. This would require a new release of Exotica to resolve.

christian-rauch, 2019-05-01:
@wxmerkt Thanks for the release.
Can you point me to these warnings? I could only see the `pinocchio` related warnings in the Jenkins job.

wxmerkt, 2019-05-01:
The deprecated/uninitialized happen during the Pinocchio build, which went through fine on the buildfarm last night. The warning in exotica has been removed - it was triggered from CMake on purpose and is now degraded to a status message.

wxmerkt, 2019-05-03:
Given the email just now, turns out that the unstable build is due to the documentation:
```
# BEGIN SECTION: Look for warnings
03:51:26 Found 14 warnings:
03:51:26 
03:51:26 -   cwd is /tmp/ws/src/exotica/exotica//tmp/ws/src/exotica/exotica/doc/Python-API.rst:11: WARNING: autodoc: failed to import module u'pyexotica'; the following exception was raised:
03:51:26 
03:51:26 - /tmp/ws/src/exotica/exotica/doc/Python-API.rst:17: WARNING: autodoc: failed to import module u'exotica_ompl_solver_py'; the following exception was raised:
03:51:26 
03:51:26 - /tmp/ws/src/exotica/exotica/doc/Python-API.rst:22: WARNING: autodoc: failed to import module u'exotica_core_task_maps_py'; the following exception was raised:
03:51:26 
03:51:26 - /tmp/ws/src/exotica/exotica/doc/Styleguide-and-Naming-Convention.rst:23: WARNING: Inline emphasis start-string without end-string.
03:51:26 
03:51:26 - /tmp/ws/src/exotica/exotica/doc/Styleguide-and-Naming-Convention.rst:23: WARNING: Inline emphasis start-string without end-string.
03:51:26 
03:51:26 - /tmp/ws/src/exotica/exotica/doc/Task_maps.rst:145: WARNING: duplicate label c++, other instance in /tmp/ws/src/exotica/exotica/doc/Styleguide-and-Naming-Convention.rst
03:51:26 
03:51:26 - /tmp/ws/src/exotica/exotica/doc/overview.rst:55: WARNING: duplicate label planning scene, other instance in /tmp/ws/src/exotica/exotica/doc/XML.rst
03:51:26 
03:51:26 - /tmp/ws/src/exotica/exotica/doc/overview.rst:115: WARNING: duplicate label task maps, other instance in /tmp/ws/src/exotica/exotica/doc/Task_maps.rst
03:51:26 
03:51:26 - /tmp/ws/src/exotica/exotica/doc/quickstart_python.rst:109: WARNING: duplicate label expected output, other instance in /tmp/ws/src/exotica/exotica/doc/quickstart_cpp.rst
03:51:26 
03:51:26 - /tmp/ws/src/exotica/exotica/doc/task_maps/com.rst:4: WARNING: duplicate label com, other instance in /tmp/ws/src/exotica/exotica/doc/Task_maps.rst
03:51:26 
03:51:26 - /tmp/ws/src/exotica/exotica/doc/using_exotica_cpp.rst:59: WARNING: duplicate label setting start position, other instance in /tmp/ws/src/exotica/exotica/doc/quickstart_python.rst
03:51:26 
03:51:26 - /tmp/ws/src/exotica/exotica/doc/Initializers.rst:95: WARNING: Could not lex literal_block as "cmake". Highlighting skipped.
03:51:26 
03:51:26 - /tmp/ws/src/exotica/exotica/doc/manual_initialization.rst:190: WARNING: Could not lex literal_block as "xml". Highlighting skipped.
03:51:26 
03:51:26 - WARNING: html_extra_path entry u'/tmp/ws/src/exotica/exotica/doc/doxygen_cpp' does not exist
03:51:26 
03:51:26 Marking build as unstable
03:51:26 # END SECTION
```

christian-rauch, 2019-05-03:
We are getting multiple unstable mails. The first was melodic development (`Mdev`) and the recent one is from the melodic documentation (`Mdoc`). The `Mdev` should be solved by the new `pinocchio_catkin` release.

wxmerkt, 2019-05-03:
20 warnings fixed, 2 remaining (from Pinocchio):
[image]

```

model.hpp:158 | Normal | GNU C Compiler 4 (gcc) | Warning:deprecated-declarations
-- | -- | -- | --
model.hpp:166 | Normal | GNU C Compiler 4 (gcc) | Warning:deprecated-declarations

```

christian-rauch, 2019-05-06:
It's not over yet.
The [`Mdoc` build job](http://build.ros.org/job/Mdoc__exotica__ubuntu_bionic_amd64/18/consoleFull) still has warnings:
> ```
> 11:49:48 sphinx-building exotica [sphinx-build -a -E -b html . /tmp/generated_documentation/api_rosdoc/exotica/html/.]
> 11:49:48   cwd is /tmp/ws/src/exotica/exotica//tmp/ws/src/exotica/exotica/doc/Python-API.rst:11: WARNING: autodoc: failed to import module u'pyexotica'; the following exception was raised:
> 11:49:49 Traceback (most recent call last):
> 11:49:49   File "/usr/lib/python2.7/dist-packages/sphinx/ext/autodoc.py", line 658, in import_object
> 11:49:49     __import__(self.modname)
> 11:49:49   File "/tmp/ws/install_isolated/lib/python2.7/dist-packages/pyexotica/__init__.py", line 3, in <module>
> 11:49:49     from ._pyexotica import *
> 11:49:49 ImportError: No module named _pyexotica
> 11:49:49 /tmp/ws/src/exotica/exotica/doc/Python-API.rst:17: WARNING: autodoc: failed to import module u'exotica_ompl_solver_py'; the following exception was raised:
> 11:49:49 Traceback (most recent call last):
> 11:49:49   File "/usr/lib/python2.7/dist-packages/sphinx/ext/autodoc.py", line 658, in import_object
> 11:49:49     __import__(self.modname)
> 11:49:49 ImportError: No module named exotica_ompl_solver_py
> 11:49:49 /tmp/ws/src/exotica/exotica/doc/Python-API.rst:22: WARNING: autodoc: failed to import module u'exotica_core_task_maps_py'; the following exception was raised:
> 11:49:49 Traceback (most recent call last):
> 11:49:49   File "/usr/lib/python2.7/dist-packages/sphinx/ext/autodoc.py", line 658, in import_object
> 11:49:49     __import__(self.modname)
> 11:49:49 ImportError: No module named exotica_core_task_maps_py
> 11:49:51 WARNING: html_extra_path entry u'/tmp/ws/src/exotica/exotica/doc/doxygen_cpp' does not exist
> ```

wxmerkt, 2019-05-06:
Ah. Those won't go away - the ROS buildfarm does not have assets built when building documentation. Yet, in order to auto-document Python we need assets built. It also complains that C++ doxygen hasn't been copied over, that won't go away either. Perhaps you know of a workaround?

christian-rauch, 2019-05-06:
I see. The modules `_pyexotica`, `exotica_ompl_solver_py` and `exotica_core_task_maps_py` are built via pybind11 and therefore cannot be parsed by sphinx before the build step.
Could we add empty placeholder modules as a workaround?

christian-rauch, 2019-05-06:
Would this help? [Generating documentation using Sphinx](https://pybind11.readthedocs.io/en/stable/advanced/misc.html#generating-documentation-using-sphinx):
> Sphinx [2] has the ability to inspect the signatures and documentation strings in pybind11-based extension modules to automatically generate beautiful documentation in a variety formats.

wxmerkt, 2019-05-06:
We are exactly using the latter - that auto-documentation feature you are quoting. However, the ROS buildfarm does not build the project before building the documentation - that job is separate and static (which is why the auto-inspection fails).

I am not convinced a placeholder is a good way to go. Adding a ``try/except`` on the import for `_pyexotica` with a warning displayed could be. That won't resolve the doxygen error though and the OMPL and Taskmap bindings would still have to be handled with a wrapper (which I'd prefer not to). For now I am just ignoring and deleting the warnings emails on the documentation job.

christian-rauch, 2019-05-06:
Well, then I will just create a filter rule after all :-)

commit 8d8ecb67 (2019-05-03): [exotica] Fix duplicate labels, resolves #547
PR #551 (2019-05-03): [exotica_pinocchio_dynamics_solver] Turn off deprecated-declarations for Pinocchio
Resolves #547 by turning off deprecated-declarations for the Pinocchio includes. Pinocchio deprecated neutralConfigurations but still uses it in the model - i.e., we are seeing warnings leading to unstable builds from things we do not use/control. Should be removed in the future once a new Pinocchio release fully removed those deprecated features.

cc: @christian-rauch
commit 3fcd9b24 (2019-05-03): [exotica_pinocchio_dynamics_solver] Turn off deprecated-declarations for Pinocchio
- Resolves #547 by turning off deprecated-declarations for the Pinocchio includes. Pinocchio deprecated neutralConfigurations but still uses it in the model - i.e., we are seeing warnings leading to unstable builds from things we do not use/control. Should be removed in the future once a new Pinocchio release fully removed those deprecated features.
closed 2019-05-06 by christian-rauch