ipab-slmc/exotica #548: catkin_lint

christian-rauch, 2019-05-02:
[`catkin_lint`](http://fkie.github.io/catkin_lint/) is a static analyser for catkin packages. It reports common and potential issues. It could be useful to integrate this into the CI.

Do you think it makes sense to acknowledge and fix these issues?

Currently, it reports:
```
$ catkin_lint src/exotica/
exotica_aico_solver: error: catkin_package() exports package include path that is not installed
exotica_aico_solver: CMakeLists.txt(43): warning: install(DIRECTORY ... DESTINATION) is not one of the ${CATKIN_*_DESTINATION}s
exotica_collision_scene_fcl: error: catkin_package() exports package include path that is not installed
exotica_collision_scene_fcl: CMakeLists.txt(18): error: catkin_package() exports pkg-config module 'LIBFCL'
exotica_collision_scene_fcl: CMakeLists.txt(47): warning: install(DIRECTORY ... DESTINATION) is not one of the ${CATKIN_*_DESTINATION}s
exotica_collision_scene_fcl_latest: error: catkin_package() exports package include path that is not installed
exotica_collision_scene_fcl_latest: CMakeLists.txt(29): warning: install(DIRECTORY ... DESTINATION) is not one of the ${CATKIN_*_DESTINATION}s
exotica_core: error: catkin_package() exports package include path that is not installed
exotica_core: error: unconfigured build_depend on 'message_runtime'
exotica_core: CMakeLists.txt(56): error: catkin_package() exports non-package include path
exotica_core: CMakeLists.txt(73): error: add_library() uses file '/generated/version.cpp' which is not in package
exotica_core: CMakeLists.txt(121): error: missing test_depend on 'rosunit'
exotica_core: warning: package 'geometry_msgs' should be listed in catkin_package()
exotica_core: warning: package 'moveit_msgs' should be listed in catkin_package()
exotica_core: warning: package 'std_msgs' should be listed in catkin_package()
exotica_core: warning: file 'cmake/generate_initializers.py' is executable but not installed
exotica_core: CMakeLists.txt(116): warning: install(DIRECTORY ... DESTINATION) is not one of the ${CATKIN_*_DESTINATION}s
exotica_core_task_maps: error: missing build_depend on 'eigen_conversions'
exotica_core_task_maps: error: catkin_package() exports package include path that is not installed
exotica_core_task_maps: error: installed target 'exotica_core_task_maps_py' is not defined
exotica_core_task_maps: CMakeLists.txt(90): error: missing test_depend on 'rosunit'
exotica_core_task_maps: CMakeLists.txt(85): warning: install(DIRECTORY ... DESTINATION) is not one of the ${CATKIN_*_DESTINATION}s
exotica_examples: error: unconfigured build_depend on 'exotica_python'
exotica_examples: error: unconfigured build_depend on 'exotica_aico_solver'
exotica_examples: error: unconfigured build_depend on 'exotica_ik_solver'
exotica_examples: error: unconfigured build_depend on 'exotica_core_task_maps'
exotica_examples: CMakeLists.txt(51): error: duplicate find_package(catkin)
exotica_examples: CMakeLists.txt(51): error: duplicate find_package(exotica_core)
exotica_examples: CMakeLists.txt(51): error: duplicate find_package(sensor_msgs)
exotica_examples: CMakeLists.txt(51): error: catkin_package() is called before find_package()
exotica_examples: CMakeLists.txt(57): error: missing test_depend on 'rosunit'
exotica_examples: CMakeLists.txt(61): error: missing test_depend on 'rosunit'
exotica_examples: warning: package 'sensor_msgs' should be listed in catkin_package()
exotica_examples: warning: file 'scripts/example_prm' is executable but not installed
exotica_examples: warning: file 'scripts/example_aico_eff_velocity' is executable but not installed
exotica_examples: warning: file 'scripts/example_ik_valkyrie' is executable but not installed
exotica_examples: warning: file 'scripts/example_aico_noros' is executable but not installed
exotica_examples: warning: file 'scripts/example_ik_interactive_cost_tuning' is executable but not installed
exotica_examples: warning: file 'scripts/run_examples' is executable but not installed
exotica_examples: warning: file 'scripts/example_fk' is executable but not installed
exotica_examples: warning: file 'scripts/example_ik_trajectory' is executable but not installed
exotica_examples: warning: file 'scripts/example_manipulability' is executable but not installed
exotica_examples: warning: file 'scripts/example_ompl_projections' is executable but not installed
exotica_examples: warning: file 'scripts/example_ik' is executable but not installed
exotica_examples: warning: file 'scripts/example_dynamic_time_indexed_problem' is executable but not installed
exotica_examples: warning: file 'scripts/example_ik_levenberg_marquardt' is executable but not installed
exotica_examples: warning: file 'scripts/example_ompl_freebase_noros' is executable but not installed
exotica_examples: warning: file 'scripts/example_ik_point_to_line' is executable but not installed
exotica_examples: warning: file 'scripts/example_ik_with_jnt_smoothing' is executable but not installed
exotica_examples: warning: file 'scripts/example_dynamic_time_indexed_problem_quadrotor' is executable but not installed
exotica_examples: warning: file 'scripts/example_ik_noros' is executable but not installed
exotica_examples: warning: file 'scripts/example_minimal' is executable but not installed
exotica_examples: warning: file 'scripts/example_ompl' is executable but not installed
exotica_examples: warning: file 'scripts/example_time_indexed_sampling' is executable but not installed
exotica_examples: warning: file 'scripts/example_shapes' is executable but not installed
exotica_examples: warning: file 'scripts/example_ompl_freebase' is executable but not installed
exotica_examples: warning: file 'scripts/example_ik_eff_axis_alignment' is executable but not installed
exotica_examples: warning: file 'scripts/example_bayesian_ik' is executable but not installed
exotica_examples: warning: file 'scripts/example_ik_look_at' is executable but not installed
exotica_examples: warning: file 'scripts/example_time_indexed_sampling_freebase' is executable but not installed
exotica_examples: warning: file 'scripts/example_aico_trajectory' is executable but not installed
exotica_examples: warning: file 'scripts/example_attach' is executable but not installed
exotica_examples: warning: file 'scripts/example_aico_joint_velocity_limit' is executable but not installed
exotica_examples: warning: file 'scripts/example_collision_distance' is executable but not installed
exotica_examples: warning: file 'scripts/example_aico' is executable but not installed
exotica_examples: warning: file 'scripts/example_lazy_prm' is executable but not installed
exotica_examples: warning: file 'scripts/example_sphere_collision' is executable but not installed
exotica_examples: warning: file 'scripts/example_ompl_noros' is executable but not installed
exotica_examples: warning: file 'scripts/example_ik_python_solver_constrained' is executable but not installed
exotica_examples: CMakeLists.txt(47): warning: malformed argument list: option 'PROGRAMS' has empty, unquoted argument
exotica_ompl_solver: error: installed target 'exotica_ompl_solver_py' is not defined
exotica_ompl_solver: CMakeLists.txt(11): error: catkin_package() exports non-package include path
exotica_ompl_solver: CMakeLists.txt(36): warning: use of link_directories() is strongly discouraged
exotica_pinocchio_dynamics_solver: error: missing build_export_depend on 'pinocchio'
exotica_python: error: installed target '_pyexotica' is not defined
exotica_python: warning: unconfigured build_depend on 'shape_msgs'
exotica_python: warning: unconfigured build_depend on 'moveit_msgs'
exotica_python: warning: unconfigured build_depend on 'geometry_msgs'
exotica_scipy_solver: warning: unconfigured build_depend on 'exotica_core'
exotica_time_indexed_rrt_connect_solver: CMakeLists.txt(10): error: catkin_package() exports non-package include path
exotica_time_indexed_rrt_connect_solver: CMakeLists.txt(20): warning: use of link_directories() is strongly discouraged
catkin_lint: checked 16 packages and found 81 problems
catkin_lint: 33 notices have been ignored. Use -W2 to see them
```

wxmerkt, 2019-05-02:
Yes, I think it makes sense to first fix them, then add `catkin_lint` to documentation and CI. Do you mind opening a PR for the first phase?

christian-rauch, 2019-05-02:
https://github.com/ipab-slmc/exotica/pull/549

I mostly fixed dependency issues. Some issues remain. It would be nice to have them resolved but it's not critical.

PR #549 (2019-05-10): Catkin lint
Fix issues in catkin packages (Fixes #548 ).

Remaining issues are mostly related to executable python scripts that are not installed and pybind modules that are not recognised.

An explanation of the messages can be found at: https://github.com/fkie/catkin_lint/blob/master/gh-pages/docs/messages.md.

Remaining issues:
```
$ catkin_lint src/exotica/
exotica_collision_scene_fcl: CMakeLists.txt(18): error: catkin_package() exports pkg-config module 'LIBFCL'
exotica_core: CMakeLists.txt(73): error: add_library() uses file '/generated/version.cpp' which is not in package
exotica_core:  [... cut]
closed 2019-05-10 by wxmerkt