ipab-slmc/exotica #554: CMake error

shyamashi, 2019-05-09:
Hi,
Ubuntu 16.04, ROS Kinetic
Installed exotica binary and was following [creating a package](https://ipab-slmc.github.io/exotica/own_robot.html) for a UR5 robot. I get the following error 
```
 CMake Error at /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by "exotica" with any
  of the following names:

    exoticaConfig.cmake
    exotica-config.cmake

  Add the installation prefix of "exotica" to CMAKE_PREFIX_PATH or set
  "exotica_DIR" to a directory containing one of the above files.  If
  "exotica" provides a separate development package or SDK, be sure it has
  been installed.
```

My package.xml is
```
<?xml version="1.0"?>
<package format="2">
  <name>ur5_exotica</name>
  <version>5.0.0</version>
  <description>The ur5_exotica package</description>
  <maintainer email="test@todo.todo">test</maintainer>
  <license>TODO</license>
  <buildtool_depend>catkin</buildtool_depend>
  <depend>roscpp</depend>
  <depend>rospy</depend>
  <depend>exotica_python</depend>

</package>
```

and CMakeLists.txt is
```
cmake_minimum_required(VERSION 2.8.3)
project(ur5_exotica)
find_package(catkin REQUIRED COMPONENTS
  roscpp
  rospy
  exotica
)
catkin_package(
  LIBRARIES
  CATKIN_DEPENDS exotica
)

include_directories(
  ${catkin_INCLUDE_DIRS}
)
```

Any help would be much appreciated
Thanks
Ash

wxmerkt, 2019-05-09:
Hi Ash,
Thank you very much for using exotica and contacting us. Apologies that you ran into this issue - we haven't pushed the updated documentation. Here is the fix: change `exotica` to `exotica_core` in the `find_package` statement and `CATKIN_DEPENDS` - it should resolve the issue.

Please let me know if you have any other questions or issues - very happy to help.

Best wishes
Wolfgang

PS: Out of curiosity, how did you come across exotica? :-)

shyamashi, 2019-05-09:
Thanks. It worked. I am working at fairspace hub surrey.

closed 2019-05-09 by shyamashi

wxmerkt, 2019-05-09:
Cool, please let us know if you have any other questions, feedback, or run into any issues - happy to help out.

wxmerkt, 2019-05-09:
The documentation is now also up to date.

shyamashi, 2019-05-10:
[python examples](https://ipab-slmc.github.io/exotica/XML-Parsing.html) hyperlink in here dont work

wxmerkt, 2019-05-10:
Thank you for raising this - I have run a link checker and fixed all similar issues as well.