ipab-slmc/exotica #556: CI ignores test failures

christian-rauch, 2019-05-09:
When running `catkin run_tests` and `catkin_test_results` on 366619192b653cc81e5af0c19ae51c369d76b762 (master) locally, I get:
```
build/exotica_examples/test_results/exotica_examples/rostest-test_python_tests.xml: 7 tests, 0 errors, 6 failures, 0 skipped
build/exotica_examples/test_results/exotica_examples/rosunit-core.xml: 5 tests, 5 errors, 0 failures, 0 skipped
Summary: 58 tests, 5 errors, 6 failures, 0 skipped
```

The resulting test xml logs ([rostest-test_python_tests.xml.txt](https://github.com/ipab-slmc/exotica/files/3164117/rostest-test_python_tests.xml.txt), [rosunit-core.xml.txt](https://github.com/ipab-slmc/exotica/files/3164118/rosunit-core.xml.txt)) contain errors. They somehow managed to pass the CI.
The CI should be made aware of these tests or we should remove them.

wxmerkt, 2019-05-11:
I think this may be on your machine only: Can you check whether you have `exotica_val_description` installed? The tests that are indicated to fail all require the meshes (listed in test_depend). You can install them via `ros-melodic-exotica-val-description`. This resolves all python test failures. The core test failure is due to a crash in your local gn_solver.

christian-rauch, 2019-05-12:
You are right. I built exotica in its dedicated workspace but I forgot that another workspace was sourced in my bashrc. The resulting errors are likely caused by the ABI incompatibility.

closed 2019-05-12 by christian-rauch