ipab-slmc/exotica #559: Not able to load robot description

shyamashi, 2019-05-10:
Hi
Ubuntu 16.04, ROS Kinetic
I was creating a package for ur5 robot for drawing the figure 8 as shown in examples. The changes I made are 
1. added URDF and SRDF for ur5 in python_ik.launch and example_ik.xml
2. EndEff frame name changed to 'ee_link' and JointGroup to 'manipulator' in example_ik.xml
3. UR5 is 6DOF hence changed StartState, NominalState, W to have 6 values in example_ik.xml and example_ik.py
4. Expected the ur5 robot to draw fig 8 when running ``` roslaunch ur5_exotica python_ik.launch ```
Instead got 
[image]
In the terminal I could see ``` [ INFO] [1557483906.181485148]: Loading robot model 'lwr'...```
Is there somewhere the robot description is hard coded? How to solve this?
Am I missing something?

Ash

wxmerkt, 2019-05-10:
Hi Ash,
Based on your description you took the correct steps. Could you share the example code so we can take a look?

Thanks,
Wolfgang

shyamashi, 2019-05-10:
Please take a look at [My github](https://github.com/shyamashi/ur5_exotica). It is almost a replica of exotica package

wxmerkt, 2019-05-10:
Thanks for sharing the repository. The issue is two-fold:
(a) your examples archive contains both the `example_ik` (without file extension, which is the original LWR example) and your new `example_ik.py` (with file extension). Since the launch file is looking for the `example_ik`, it picks up the LWR. I deleted all other copy-pasted file and changed the launch file to `example_ik.py to fix this issue.
(b) The UR5 URDF does have world_joint in the URDF - which is a semantic and not a robot joint. This is an error in the URDF. Removing it fixes it.

You can check the necessary changes to get your example to run here: https://github.com/wxmerkt/ur5_exotica/commit/133b543010eb37360dfbe7848b8363d12d3789b1. I've also PR'd it to your repository (https://github.com/shyamashi/ur5_exotica/pull/1/).

shyamashi, 2019-05-10:
Thanks for the reply. 
Still drawing figure 8 does not work when I run ``` roslaunch ur5_exotica python_ik.launch ```. Still get the  
[image]
but in the terminal it shows loading robot model ur5

wxmerkt, 2019-05-10:
You need to change the fixed frame on the left hand side (in RViz) to the one specified in the SRDF as virtual joint. I think it was fixed_base? If so it’s “exotica/fixed_base”

shyamashi, 2019-05-10:
sorry that was a stupid mistake. Thanks. It works. One more issue is that it does not recognize the ground plane in Rviz and goes under it to draw the figure 8

wxmerkt, 2019-05-10:
Happy to hear!

This is expected: RViz does send any information on environment etc back (the targets are relative to the base link and those go below the zero mount - you can adjust those in the XML for frames/the trajectory (for the trajectory based examples)/the set position (figure eight). The basic IK example does not include collision avoidance either.

closed 2019-05-10 by wxmerkt