ipab-slmc/exotica #568: error at string construction in create_problem

christian-rauch, 2019-05-16:
My python program exits with message: `basic_string::_M_construct null not valid`.

There is no backtrace. gdb reports that the process exited normally. The message comes from a `std::logic_error` and is thrown when constructing a string from `NULL` or `nullptr`. I cannot see where this exception is caught in the code and I therefore cannot retrace why only the message is shown without the error `terminate called after throwing an instance of`.

This occurred to me after switching the problem type from `UnconstrainedEndPoseProblem` to `SamplingProblem`. E.g. in the snippet:
```Python
print("problem...")
print("conf:", self.prob_conf)
self.problem = exo.Setup.create_problem(self.prob_conf)
print("...done")
```
the process never reaches `...done`.

This configuration works:
`('conf:', ('exotica/UnconstrainedEndPoseProblem', {'PlanningScene': [('exotica/Scene', {'URDF': '[...]/kinova_description/urdf/j2n6s300.urdf', 'SRDF': '[...]/kinopt/cfg/srdf/j2n6s300.srdf', 'JointGroup': 'all'})], 'Maps': [], 'Cost': [], 'Name': 'problem'}))`

This fails:
`('conf:', ('exotica/SamplingProblem', {'PlanningScene': [('exotica/Scene', {'URDF': '[...]/kinova_description/urdf/j2n6s300.urdf', 'SRDF': '[...]/kinopt/cfg/srdf/j2n6s300.srdf', 'JointGroup': 'all'})], 'Maps': [], 'Cost': [], 'Name': 'problem'}))`

The process exits before reaching `return Setup::CreateProblem(init);` in [`setup.def_static("create_problem", [](const Initializer& init)`](https://github.com/ipab-slmc/exotica/blob/0a8f969409fecc2d63db92c7390ad4dbd511fe8c/exotica_python/src/pyexotica.cpp#L498).

I, therefore, assume that it is related to the initialiser.

christian-rauch, 2019-05-16:
It must be the initialiser.

The same happens when calling `create_solver` with `exotica/SamplingProblem`. I know this does not make sense but it suggests that the issue is related to the initialiser.

Any hints where a string is created from a raw pointer in the initialisers?

christian-rauch, 2019-05-16:
This python script reproduces the issue: [example_rrt_sampling_p2l.py.txt](https://github.com/ipab-slmc/exotica/files/3189080/example_rrt_sampling_p2l.py.txt).

wxmerkt, 2019-05-17:
I found the issue: You are trying to specify a `Cost`, when `SamplingProblem` does not take a cost. Remove the `Cost` Initialiser in the problem (or rename to `Equality` or `Inequality`) and your script passes.

christian-rauch, 2019-05-17:
Thanks, indeed.
But, shouldn't the initialiser ignore unused parameters?
I do this quite often with solvers. I define the super-set of parameters in the python dict and the configured solver picks the parameters that are defined in the initialiser file.

Anyway, this silent exit is no valid behaviour.
The initialiser should either (1) ignore the unused parameter, or (2) throw a proper error with a hint how to fix this.

wxmerkt, 2019-05-17:
I agree with you and I've noticed this inconsistent behavior in the past - it may be related to whether the element is a simple `Optional` (`double`) or a vector of `Initialiser` itself. Very happy if you find the source and an easy fix :)

christian-rauch, 2019-05-17:
That's a good point. This ignoring works with POD types but I assume that `Initializer`s are stored as a pointer, which then might be NULL.

christian-rauch, 2019-05-17:
Thanks to the awesome [ROS plugin for Qt Creator](https://github.com/ros-industrial/ros_qtc_plugin) I was able to debug and fix the issue. (I am a big fan of this tool :smile: )

The error is caused when creating the string here:
https://github.com/ipab-slmc/exotica/blob/0a8f969409fecc2d63db92c7390ad4dbd511fe8c/exotica_python/src/pyexotica.cpp#L65

`char* PyString_AsString(PyObject *string)` can returns a NULL pointer if:

> If string is not a string object at all, PyString_AsString() returns NULL and raises TypeError.

And this triggered because a parameter that is not found is converted to a string here:
https://github.com/ipab-slmc/exotica/blob/0a8f969409fecc2d63db92c7390ad4dbd511fe8c/exotica_python/src/pyexotica.cpp#L379-L382
What's the purpose of this?

wxmerkt, 2019-05-17:
Fixed in #566

closed 2019-05-17 by wxmerkt