ipab-slmc/exotica #569: SamplingProblem scalar costs

christian-rauch, 2019-05-17:
The scalar costs for an end pose problem is defined as:
`y_d^T * S * y_d`
i.e. the sum of squares of individual elements of `y_d` weighted by `s`.

The `SamplingProblem` defines `inequality` and `equality`, each with their own `y_d` and `S`.

How would you weight those to get a scalar cost for a sampling problem? Do you sum or average them?

wxmerkt, 2019-05-17:
`SamplingProblem` as `inequality` and `equality` to define validity in `IsValid` (together with tolerances that you can set as for random samples to lie on the constraint manifold is unlikely). We explicitly do not yet have a scalar cost (which would be needed for say `RRT*` with a custom objective instead of just path-length).

We could consider adding `Cost` if it defaults to path-length if not defined and is correctly passed into the OMPL solvers. In that case, just do the same as for the scalar costs in an end-pose problem. I would not use inequality/equality to arrive at that cost function though but a dedicated `Cost` initializer. If you really wanted to - use a square of the positive only terms on the inequality and a squared term on the equality.

christian-rauch, 2019-05-17:
I am not sure if I get why there is no `cost` in `SamplingProblem`.

The `EndPoseProblem` has `inequality`, `equality` and `cost`. It minimises an objective with constraints using the Jacobian. The `UnconstrainedEndPoseProblem` does the same without constraints.

Why can the `SamplingProblem` not have a `SamplingTask cost;` that represents `y_d` at the sampled states? Should there be a `UnconstrainedSamplingProblem` that has this `cost` without constraints?

This confusion of mine might be because the `SamplingProblem` was used in a different context than what I am looking into:
For a classical particle swarm type of optimiser, I only need to evaluate the samples. I can already do this with the `UnconstrainedEndPoseProblem` by ignoring the gradients/Jacobian. But since the `UnconstrainedEndPoseProblem` expects that gradients are used by the solver, the Jacobian is computed numerically calling the objective function once more. For a more efficient optimisation, I want to skip this gradient computation. A workaround is to override the `Update` method that returns the Jacobian and just do not return a Jacobian.
I am therefore basically looking for a `SamplingProblem` that does not use constraints or a `UnconstrainedEndPoseProblem` that does not use gradients.

wxmerkt, 2019-05-17:
> For a more efficient optimisation, I want to skip this gradient computation.

You can set `DerivativeOrder` in any PlanningProblem to be 0 (no Jacobians!). I use that for derivative-free evolutionary and black-box optimisation. I think what you want is `UnconstrainedEndPoseProblem` with `DerivativeOrder=0`.

christian-rauch, 2019-05-17:
That also did the trick, thanks. I thought that the `flags_` are fixed per problem type and cannot be changed at runtime.

closed 2019-05-17 by christian-rauch

wxmerkt, 2019-05-17:
From what I recall they cannot be changed at runtime - but they can be changed during creation.

christian-rauch, 2019-05-17:
With runtime, I meant that I thought they are set during compilation and cannot be changed afterward at runtime of the process via the initialisers.