ipab-slmc/exotica #587: Avoid errors at runtime that could be caught at compilation time.

cmower, 2019-05-31:
* This problem really stems from my own naive knowledge of C++ however I thought I'd post and ask.
* I solved some of the issues being discussed in #581 and worked out I didn't need to override `AssignScene`. I forgot to remove the definition in the `.h` file but the code built just fine.
* When I came to run the example script the following error was thrown:
```
Traceback (most recent call last):
  File "/home/chris/Projects/kuka_ws/src/exotica/exotica/exotica_examples/scripts/example_ik_avoid_look_at_sphere", line 3, in <module>
    import pyexotica as exo
  File "/home/chris/Projects/kuka_ws/devel/lib/python2.7/dist-packages/pyexotica/__init__.py", line 35, in <module>
    exec(__fh.read())
  File "<string>", line 3, in <module>
ImportError: Failed to load library /home/chris/Projects/kuka_ws/devel/lib//libexotica_core_task_maps.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = /home/chris/Projects/kuka_ws/devel/lib//libexotica_core_task_maps.so: undefined symbol: _ZN7exotica17AvoidLookAtSphere11AssignSceneESt10shared_ptrINS_5SceneEE)
terminate called after throwing an instance of 'class_loader::LibraryUnloadException'
  what():  Attempt to unload library that class_loader is unaware of.
[example_ik_avoid_look_at_sphere_node-2] process has died [pid 10991, exit code -6, cmd /home/chris/Projects/kuka_ws/src/exotica/exotica/exotica_examples/scripts/example_ik_avoid_look_at_sphere __name:=example_ik_avoid_look_at_sphere_node __log:=/home/chris/.ros/log/47ca390c-83b7-11e9-8de6-a0cec809263b/example_ik_avoid_look_at_sphere_node-2.log].
log file: /home/chris/.ros/log/47ca390c-83b7-11e9-8de6-a0cec809263b/example_ik_avoid_look_at_sphere_node-2*.log
```
* After a few headaches I worked out my error and solved the issue. 

I wonder if there is a way to check for this kind of issue at compilation time to avoid wasting time?

VladimirIvan, 2019-05-31:
That is up to the compiler and linker really. Most of the time, the linker will complain that the method hasn't been implemented. It may have been just an artifact of lazy checking or special tratment because it was a virtual method that it didn't. StackOverflow may have the answer. I wouldn't be surprised if there is a warning/error flag that you can enable for GCC.

The python wrapper will often complain about these type of things because it creates a lot of the classes when the pyexotica is loaded to set up the reflection.

I don't have a real answer beyond complaining about c++. We'll leave that for the pub though.

wxmerkt, 2020-02-28:
Closing as some of these issues that are beyond ABI incompatibility [which is the error above] have been fixed by #693.

closed 2020-02-28 by wxmerkt