ipab-slmc/exotica #593: exotica_python: Numpy array vs matrix

wxmerkt, 2019-06-09:
Our Python wrapper returns ``numpy.array``. However, many times that breaks the shape of vectors or just makes it hard to get the correct matrix operations (it's easy to end up with incorrect operations). 

It would be better to return ``numpy.matrix``. On a quick search I haven't found a solution with pybind11 yet, but we should investigate this in the future.

jcarpent, 2019-06-24:
[EigenPy](https://github.com/stack-of-tasks/eigenpy) may help you at this stage. We support both, and you can switch the conversion online.

wxmerkt, 2019-06-24:
Thank you Justin :-) We'd like to retain pybind11 and avoid boost-python. I didn't see a way to do this with EigenPy, but may have missed it?

jcarpent, 2019-06-25:
I do understand your choice. In our context, we use the following lines to perform the switch:
```
import eigenpy
eigenpy.switchToNumpyArray()
```

wxmerkt, 2019-06-25:
[`np.asmatrix`](https://docs.scipy.org/doc/numpy/reference/generated/numpy.asmatrix.html) is the solution of choice (which does not make a copy) until we find a way to switch within pybind11, which likely requires a fix to it.

closed 2019-10-14 by wxmerkt