ipab-slmc/exotica #604: Load XML should convert to arrays in xml to numpy arrays rather than strings

cmower, 2019-06-19:
This issue is seen easier by a simple example. I have a problem in `test.xml` that is 
```xml
<?xml version="1.0" ?>
<TestConfig>

  <EndPoseProblem>

    <StartState>1 2 3</StartState>
    
  </EndPoseProblem>
  
</TestConfig>
```
In python, when I load the initializer and inspect the `StartState` I get the following
```python
>>> pr = exo.Initializers.load_xml('/path/to/test.xml')
>>> pr
('exotica/EndPoseProblem', {'StartState': u'1 2 3'})
>>> pr[1]['StartState']
u'1 2 3'
>>> type(pr[1]['StartState'])
unicode
```
However, if I initialize from `exo` and inspect start state as follows.
```python
>>> pr = exo.Initializers.EndPoseProblemInitializer()
>>> pr[1]['StartState']
array([], dtype=float64)
>>> type(pr[1]['StartState'])
numpy.ndarray
```
This is creating problems for me when I want to start with a base xml and re-initialize later on. There are hacks round this but I feel that a more elegant solution should exist. That is, when `exo.Initializers.load_xml` is called then `pr[1]['StartState']` (and similar parameters) should be passed as `numpy.ndarray`'s rather than `unicode`. My guess is that this issue is coming from the handling of casting data types between C++ and Python in `pyexotica.cpp` but I am not sure how to fix?

wxmerkt, 2019-06-19:
The difference is that `load_xml` returns a generic `Initializer`. While you create the specific initializer in Python. The parsing into the different categories (e.g., double or array) only happens when you create the specific initializer. 

We now need to replicate this process in Python - and it may not be exposed yet (dont know off the top of my head):

```
generic_initializer = exo.Initializers.load_xml('/path/to/test.xml')
pr = exo.Initializers.EndPoseProblemInitializer(generic_initializer)
```

(Reference: https://github.com/ipab-slmc/exotica/blob/cad9fbad9906cefed16cf0d16fbe05dc9c163a23/exotica_core/cmake/generate_initializers.py#L199)

VladimirIvan, 2019-06-20:
You can run `pr = exo.Initializers.Initializer(exo.Initializers.load_xml('/path/to/test.xml'))`.
That will parse the initializer into the specialized type in c++ and then return it to python as a dict with the correct data types.

wxmerkt, 2019-06-21:
Closing this issue as it was a non-issue. If still unclear, please reopen.

closed 2019-06-21 by wxmerkt