ipab-slmc/exotica #606: FK dead link

christian-rauch, 2019-06-24:
I am encountering a dead link in FK (`The pointer to KinematicElement A is dead.`) on a link that I manually added via `GetKinematicTree().AddElement(camera_frame, cam_pose, root);` where `camera_frame` and `root` are of type `std::string`.

Right after adding the link, the returned pointer is not `nullptr` and I can access the link via `GetKinematicTree().GetTreeMap().at(camera_frame).lock()->segment.getName()`.

Also inside FK just before the pointer check, I can see that `tree_map_` contains the added frame (i.e. `second` points to a valid element) but the provided element_A is `nullptr`.

Under what circumstances can `lock()` give me a `nullptr`?

christian-rauch, 2019-06-24:
Appearently, you need to store the returned `shared_ptr` like:
`this->camera_element = this->scene_->GetKinematicTree().AddElement(...)`
to not have the `weak_ptr` expire.

> If the object is weak_ptr::expired, a default-constructed shared_ptr object.

The `KinematicElement` does not have a default constructor. I assume in this case, a `nullptr` is returned.

Is it intended that the pointer returned by `AddElement` must be stored?

wxmerkt, 2019-06-24:
> Is it intended that the pointer returned by AddElement must be stored?

Yes, definitely you have to store it. We store it in different maps, e.g., `environment_tree_`. We also use `AddEnvironmentElement`, which as far as I can remember, automatically stores it. Some of these AddElement methods are only exposed for the sake of the scene creation and probably should be slightly redesigned from an API standpoint.

christian-rauch, 2019-06-26:
As demonstrated in the example at: https://en.cppreference.com/w/cpp/memory/weak_ptr
the `weak_ptr` expires when the original `shared_ptr` goes out of scope. I assume this situation would not appear, if `tree_` would store shared instead of weak pointers.

Why does `tree_` store `weak_ptr` instead of `shared_ptr`? I.e. shared pointers would always maintain their use_count and stay alive until the last object goes out of scope. I imagine there are more of these situations hidden in the code. This is especially problematic with objects that cannot be default constructed.

wxmerkt, 2019-06-26:
The answer is you should _not_ use `AddElement` but `AddEnvironmentElement`, or take care of storing the pointer yourself. `tree_` used to store `shared_ptr` but shouldn't have - it lead to memory issues. We only keep one original copy and use `weak_ptr` in `tree_`.

We use several trees:
- `model_tree_` => Only the robot model frames, created on initialisation
- `tree_` => Gets reset on every `Scene::UpdateInternalFrames` and recreated from `model_tree_` + auxiliary frames
- `environment_tree_` => Auxiliary frames added later, such as the ones you are trying to add

christian-rauch, 2019-06-26:
Right, `AddEnvironmentElement` uses `AddElement` to add the frames and stores the returned pointer in `environment_tree_`. Maybe the `AddElement` methods should become private.

closed 2019-06-26 by christian-rauch

wxmerkt, 2019-06-26:
Yes, that is what I meant by saying we may need to reconsider that API at some point in the future. However, `Scene` still uses it for some of the custom links to re-create upon scene creation (which could be pushed into `KinematicTree` at some point).