ipab-slmc/exotica #613: Non-deterministic roll-outs in iLQG example

wxmerkt, 2019-06-28:
Running `rosrun exotica_examples example_dynamic_time_indexed 10` results in vastly different (read unstable) roll-outs despite using the feedback gains - the first two are essentially the same, the third is... unstable.

include4eto, 2019-06-30:
Why is this an issue? Isn't that what (velocity) noise is supposed to look like? I'll look into it to make sure there aren't any mistakes, but it looks OK to me. Note I'm not seeding the generator, so it should always be the same (random) roll-outs.

wxmerkt, 2019-06-30:
When using the feedback control policy, I do expect some determinism - or is the noise perhaps set too large?

include4eto, 2019-08-19:
https://github.com/ipab-slmc/exotica/pull/656

closed 2019-09-05 by wxmerkt