ipab-slmc/exotica #626: global seed KinematicTree

christian-rauch, 2019-07-09:
I added a `KinematicTree::SetSeed` method in https://github.com/ipab-slmc/exotica/pull/487 to allow reproducible sampling of configurations.

A call to `KinematicTree::BuildTree` resets the random number generator: https://github.com/ipab-slmc/exotica/blob/7905a390e1b88ef58c7ad4f418983bdf303b21d5/exotica_core/src/kinematic_tree.cpp#L283 and requires to set the seed again.

Q1: Is this reset of the `generator_` required? Can it be removed or should we replace `rd_()` by the user set seed?

I think it would be good if EXOTica would keep a global seed for all random number generator that may be used in solvers, tasks, etc., to enable reproducible results and easier debugging. This seed would be determined once at the beginning and all random number generators would need to use the same seed, e.g. an `exotica::seed()` that either returns a random or static user chosen seed.

Q2: Does this sound feasible? What would be a good approach to make sure that all random number generators use the same seed?

wxmerkt, 2019-07-09:
> A call to KinematicTree::BuildTree resets the random number generator:

A call to `KinematicTree::BuildTree` happens only on `Instantiate`, which happens only when you create a new Scene, which happens only when you create a problem. As thus, does it really reset? [I may be missing something here]

christian-rauch, 2019-07-10:
Indeed, I create the problem more than once. Sometimes I do this to obtain properties of the kinematic tree, before configuring the actual task. As a workaround, you could reset the seed every time the problem is created.

However, I think in my particular problem the issue is that the seed for the `KinematicTree` is only set after `Setup::CreateProblem` has been called. Solvers, etc. that are `Instantiate`d during `CreateProblem` are using an unseeded `KinematicTree`. I am working around this by providing the seed manually to the initialiser and seeding `KinematicTree` before using `KinematicTree::SetSeed`.

This does not scale well, if you have many more random number generators in exotica. I am therefore wondering if EXOTica should have a global seed that is determined when it is instantiated. This global-seed would then be used to draw new seeds for any generator in EXOTica. This way, you can reset generators, but deterministically.

wxmerkt, 2019-07-10:
Makes sense - where would you suggest to place it?

christian-rauch, 2019-07-10:
Since there are many entry points to initialise exotica, I am not sure about this. It appears to me that `CreateSolver` is always called first, i.e. the `Setup` class should store that seed and make it available via a getter `Setup::GetSeed()`.
From a quick search, `std::random_device` is only used in three places: `Renderable`, `KinematicTree` and `RandomColor`. Do you have an idea where randomness is used elsewhere?

wxmerkt, 2019-07-10:
OMPL solvers. I think `Setup` may be the best place to store. You can also create Problems and Scenes without a Solver, so Setup is the place to go.