ipab-slmc/exotica #627: Jacobian computation frame B offset ignored

christian-rauch, 2019-07-10:
I used `scene_->GetKinematicTree().Jacobian()` to compute Jacobians inside task maps.
As I understand, `Jacobian(element_A, offset_a, element_B, offset_b)` should give you the gradients that minimise position `offset_a` in `element_A` with respect to `offset_b` in `element_B`.

However, in this example (just added to `PointToLine::Update` and run `test_maps`):
```C++
std::cout << "P2L J0:" << std::endl << scene_->GetKinematicTree().Jacobian("endeff", KDL::Frame(KDL::Vector(0,0,0)), "base", KDL::Frame(KDL::Vector(0,0,0))) << std::endl;
std::cout << "P2L J1:" << std::endl << scene_->GetKinematicTree().Jacobian("endeff", KDL::Frame(KDL::Vector(0,0,0)), "base", KDL::Frame(KDL::Vector(0.5,0.5,0.5))) << std::endl;
std::cout << "P2L J2:" << std::endl << scene_->GetKinematicTree().Jacobian("endeff", KDL::Frame(KDL::Vector(0,0,0)), "base", KDL::Frame(KDL::Vector(1,1,1))) << std::endl;
std::cout << "P2L J3:" << std::endl << scene_->GetKinematicTree().Jacobian("endeff", KDL::Frame(KDL::Vector(0.4,0.5,0.6)), "base", KDL::Frame(KDL::Vector(0,0,0))) << std::endl;
```
gives me the very same Jacobian for J0, J1, J2 with different `offset_b`. It appears as if `offset_b` is ignored.

This seems to be related to https://github.com/ipab-slmc/exotica/issues/181.

christian-rauch, 2019-07-10:
The offset is computed here:
https://github.com/ipab-slmc/exotica/blob/7905a390e1b88ef58c7ad4f418983bdf303b21d5/exotica_core/src/kinematic_tree.cpp#L765
but `tmp` is never used.

We should enable warnings to catch these issues.

christian-rauch, 2019-07-10:
In the section where the effect of frame B on the Jacobian is computed
https://github.com/ipab-slmc/exotica/blob/7905a390e1b88ef58c7ad4f418983bdf303b21d5/exotica_core/src/kinematic_tree.cpp#L777-L787 
non-controlled links like the root are basically ignored. I guess this is the reason why the `offset_b` for `"base"` does not change the Jacobian.

But I also get the same Jacobians with `"link1"` as frame B. I noticed that the reference point to which the gradient/twist is computed for B is set to `RefPoint(frame.temp_A.p - it->frame.p)`. I believe this should be `RefPoint(frame.temp_B.p - it->frame.p)` since the gradients otherwise cancel out.

Should the reference point be `temp_B.p - it->frame.p` and how can be use non-controlled frames for frame B?

christian-rauch, 2019-07-17:
The Jacobians only depend on the orientation part of the offset.
Offsets with the same orientation will result in the same Jacobians:
```C++
std::vector<std::pair<std::string, KDL::Frame>> b_offsets;
b_offsets.emplace_back("identity", KDL::Frame());
b_offsets.emplace_back("trans", KDL::Vector(0.5,0.5,0.5));
b_offsets.emplace_back("rotz", KDL::Rotation::RotZ(M_PI/4));
b_offsets.emplace_back("rotz_trans", KDL::Frame(KDL::Rotation::RotZ(M_PI/4), KDL::Vector(0.5,0.5,0.5)));

for(const auto & [name, offset] : b_offsets) {
    std::cout << name << " J:" << std::endl << scene_->GetKinematicTree().Jacobian("endeff", KDL::Frame(), "base", offset) << std::endl;
}
```
gives the same Jacobian for `identity` and `trans`, and for `rotz` and `rotz_trans`.

closed 2019-07-17 by christian-rauch

VladimirIvan, 2019-07-17:
A postmortem explanation based on a discussion in the corridor:
The Jacobian columns are twists of the end-effector produced by a unit rotation/motion of the joint expressed in the base frame. Changing the base frame is equivalent to changing the expressed-in frame. This operation only performs a multiplication by the rotation matrix of the relative transform from current expressed-in frame to the new expressed-in frame. Hence the Jacobian changes only if the rotation of the base frame changes.