ipab-slmc/exotica #634: Storing and loading feedback policies

wxmerkt, 2019-07-16:
- A `FeedbackMotionSolver` only returns the control trajectory as a solution, while the gains remain transient in that instance of the solver.
- This makes loading and replaying control policies impossible.
- The gains are currently stored by the individual solvers, and not in the central `FeedbackMotionSolver`. DDP/iLQG only have K/k and L/l respectively. iLQR has extra gains.

We need a mechanism to store and load a full policy with solution and gains. Perhaps that requires a data struct (u, K, k, x_ref). Let's have a discussion what's best to proceed. At the bare minimum, we need to be able to access K, k, and x_ref.

VladimirIvan, 2019-07-17:
Agreed. Variable type return values are long overdue.

I'm hoping for a lightweight flexible solution that the solvers can populate with different data as required.
Maybe based on maps of boost::any and python dictionaries in the wrapper. Possibly with a default conversion to an Eigen::MatrixXd/numpy.array of the state strajectory.

wxmerkt, 2019-07-18:
I'd be happy to go with a `struct` that can be retrieved manually from the solver, i.e., that isn't the `solution` returned. Otherwise your suggestion sounds great, but is part of a larger overhaul.

wxmerkt, 2021-04-08:
We've manually exposed the required quantities from the individual solvers. While a better solution with variable type returns would be nice, it's not a priority.

closed 2021-04-08 by wxmerkt